carla-simulator/carla #5557: Why does the vehicle not move when collision happend to it

KouWeiBin, 2022-07-02:


KouWeiBin, 2022-07-02:
CARLA version: 0.9.11
Platform/OS: windows 10

Hi, 
recently I build a new map into carla. now I can successfully add several walkers and a vehicle onto the map. I tried use the manual_control.py to control the vehicle, the vehicle move correctly. but when collision happened to it, no matter what I do, say, move forword, reverse, turning right/left, the vehicle can't move anymore. 

so could please anyone give me some useful information about how to fix it？ Thanks

Blyron, 2026-11-12:
There is not enough information to reproduce this, and it was on CARLA 0.9.11. Closing. If it still happens there, please open a new issue with the steps to reproduce and the logs.

closed 2026-11-12 by Blyron