carla-simulator/carla #5559: error when unload_map_layer

WindTi, 2022-07-04:
CARLA version: 0.9.12
Platform/OS: ubuntu 18.04 LTS

Problem you have experienced:
I use the python api unload_map_layer to remove the redundant elements of the map, and then the server will crash when I use scenario_runner. The problem can also be simplified into: using unload_map_layer and then adding a new client then carla server will crash

What you expected to happen:
server can work normally after using API unload_map_layer and adding a new client

felixlutz, 2022-09-03:
I have the same issue with CARLA 0.9.13 on Ubuntu 20.04.

WindTi, 2022-09-05:
> 

> I have the same issue with CARLA 0.9.13 on Ubuntu 20.04.

Using this API unload_map_layer to dynamically delete elements and run scenario_runner doesn't seem to work, because it will have a reload_world action. My solution is to directly delete the texture assets in the map file and modify it into a new map

closed 2022-09-22 by WindTi

Ksusha06, 2024-05-23:
Well, it is not a solution. Obviously, this is a bug, which should be fixed by developers.

Devs, please, fix this silly error <3