carla-simulator/carla #5560: Lidar doesn't work well in the map customized by RoadRunner

wujingda, 2022-07-04:
Hi everyone,

Recently we use the Roadrunner to generate a new map and import it into CARLA. The map works well with cameras but becomes weird when facing the lidar. Specifically, I use the open3d to visualize the lidar data, and the lidar data in this new map does not show the line (ray cast) of the ground. The lidar data in our new map (figure 1) and in an official map (figure 2) are provided for comparison. BTW, the lidar setting is invariant across maps.
Does anyone have ideas about the issue? Is it because of the road texture of the map or something... we are stuck by this issue for days. Many thanks.

CARLA version: 0.9.13
Platform/OS: Win10

[image]
[image]

TeleYuhao, 2023-03-23:
I have met similar problem , do u have solve it?

TeleYuhao, 2023-03-28:
You can try to import map through "make import ".  Sometimes the problem could be solved.

wujingda, 2023-06-01:
这是来自QQ邮箱的自动回复邮件。您好，您的邮件已经收到，谢谢。

MIKE-GUO233, 2025-11-05:
I have met similar problem , do u have solve it?

Blyron, 2026-11-12:
The cause was simple collision on the imported road meshes. The import now sets *Use Complex As Simple* collision on them; for meshes imported by hand, set Collision Complexity to *Use Complex As Simple*. Closing.

closed 2026-11-12 by Blyron