carla-simulator/carla #5562: fatal error C1083: Cannot open include file: 'proj_api.h': No such file or directory

Nozoomhs, 2022-07-04:


Tumb1eweed, 2022-07-07:
proj_api.h is deleted after 8.0.0, so you can install a older version, between 5.0.0 and 8.0.0 (5.0.0 declare this file would be deprecated)

Nozoomhs, 2022-07-19:
> proj_api.h is deleted after 8.0.0, so you can install a older version, between 5.0.0 and 8.0.0 (5.0.0 declare this file would be deprecated), you can modify to proj.h. [image]

How do I exactly do that? I could only find version after 0.8 is there a repository for that?

Tumb1eweed, 2022-07-19:
> > proj_api.h is deleted after 8.0.0, so you can install a older version, between 5.0.0 and 8.0.0 (5.0.0 declare this file would be deprecated), you can modify to proj.h. [image]
> 
> How do I exactly do that? I could only find version after 0.8 is there a repository for that?

oh i use **conan** to manage dependencies, you can try too.
https://conan.io/center/proj

Tumb1eweed, 2022-07-19:
> oh i use **conan** to manage dependencies, you can try too. https://conan.io/center/proj

There is also repo in Github: https://github.com/OSGeo/PROJ/releases?page=3

Nozoomhs, 2022-07-20:
> oh i use **conan** to manage dependencies, you can try too. https://conan.io/center/proj There is also repo in Github: https://github.com/OSGeo/PROJ/releases?page=3

Sorry I am still a little bit confused, so you are saying that the carla version that I am working with now is fine, but  I have to add proj library for it to work? If so exatcly how? Thank you once again.

Tumb1eweed, 2022-07-21:
> Sorry I am still a little bit confused, so you are saying that the carla version that I am working with now is fine, but I have to add proj library for it to work? If so exatcly how? Thank you once [...]

Yes, you can just follow the installation of proj to install following version( >= 5.0.0 < 8.0.0).

Nozoomhs, 2022-07-21:
> 

I have followed the instructions, decided to download proj 7.2.1, unzipped the tar file then according to this installation description [https://proj.org/install.html] tried to install it, since with the recommended way (OSGeo4W) I could have only installed 9.0.1 or 9.0.2. After trying to build it with "cmake -G "Visual Studio 16 2019" .."  I got the following error:
C:\proj-7.2.1\build>cmake -G "Visual Studio 16 2019" ..
-- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19042.
-- The C compiler identification is MSVC 19.29.30145.0
-- The CXX compiler identification is MSVC 19.29.30145.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Requiring C++11
-- Requiring C++11 - done
-- Requiring C99
-- Requiring C99 - done
-- Configuring PROJ:
-- PROJ_VERSION                   = 7.2.0
-- PROJ_ABI_VERSION               = 7_2
-- Looking for include file dlfcn.h
-- Looking for include file dlfcn.h - not found
-- Looking for include file inttypes.h
-- Looking for include file inttypes.h - found
-- Looking for include file memory.h
-- Looking for include file memory.h - found
-- Looking for include file stdint.h
-- Looking for include file stdint.h - found
-- Looking for include file stdlib.h
-- Looking for include file stdlib.h - found
-- Looking for include file string.h
-- Looking for include file string.h - found
-- Looking for include file sys/stat.h
-- Looking for include file sys/stat.h - found
-- Looking for include file sys/types.h
-- Looking for include file sys/types.h - found
-- Looking for include file unistd.h
-- Looking for include file unistd.h - not found
-- Looking for 4 include files stdlib.h, ..., float.h
-- Looking for 4 include files stdlib.h, ..., float.h - found
-- Looking for localeconv
-- Looking for localeconv - found
-- Looking for strerror
-- Looking for strerror - found
CMake Error at CMakeLists.txt:129 (message):
  sqlite3 binary not found!

CMake Error at CMakeLists.txt:134 (message):
  sqlite3 dependency not found!

CMake Error at CMakeLists.txt:140 (message):
  sqlite3 >= 3.11 required!

CMake Error at C:/Program Files/CMake/share/cmake-3.23/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find TIFF (missing: TIFF_LIBRARY TIFF_INCLUDE_DIR)
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.23/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  C:/Program Files/CMake/share/cmake-3.23/Modules/FindTIFF.cmake:124 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:151 (find_package)

-- Configuring incomplete, errors occurred!
See also "C:/proj-7.2.1/build/CMakeFiles/CMakeOutput.log".
See also "C:/proj-7.2.1/build/CMakeFiles/CMakeError.log".

Thank you for your help in advance again.

ishaan95, 2022-09-13:
I had a similar initial error. In the script `Util/InstallersWin/install_proj.bat`, `proj-7.2.1.tar.gz` could not be downloaded. It was returning the error `The request was aborted: Could not create SSL/TLS secure channel`. This resulted in `OSM2ODR` not installing properly. I changed the download link to the github release one by changing lines 42-46 in the script `install_proj.bat`: 

```
set PROJ_VERSION_NUMBER=7.2.1
set PROJ_BASE_NAME=proj-%PROJ_VERSION_NUMBER%
set PROJ_ZIP=%PROJ_BASE_NAME%.tar.gz
set PROJ_ZIP_DIR=%BUILD_DIR%%PROJ_ZIP%
set PROJ_REPO=https://github.com/OSGeo/PROJ/releases/download/%PROJ_VERSION_NUMBER%/%PROJ_ZIP%
```

omitheaussie, 2023-10-13:
For me in Oct 2023 (seems this is a moving target as people keep changing file names/scripts), on line 53, I had to change the line from:
`set PROJ_SRC_DIR=%BUILD_DIR%proj-src`
to
`set PROJ_SRC_DIR=%BUILD_DIR%proj-src\proj-7.2.1`