carla-simulator/carla #5566: find 3d world point from sensor pixel point

MMNavetty, 2022-07-06:
Hello everyone,
I want to make a projection of a sensor detection in a map, so i need the homography matrix to transform a pixel coordinate into 3d world point (/ 2d world point, z coord is not important).

Using client_bounding_boxes.py, i can get a sensor pixel point from a worlds point, but i can't find how to do the opposite.
I tried using the calibration matrix, and tried the get_matrix() function of object ClientSideBoundingBoxes but i can't find something that work.
Can someone help me ?
Thanks in advance,
MM

MMNavetty, 2022-07-07:
For those intersted, i'm sharing my 2 functions : 
make calibration is to get P, the "world to cam screen point " homography matrix, whereas brute_force_inversion is to get the pixel2world homography matrix. Since the inversion of a 3*4 matrix is not precise at all and doesnt seem to be working for me, that's the best technique i found so far.

`def make_calibration(sensor_manager):

    K = sensor_manager.sensor.calibration

    # P = K * F * [R t]
    flip = np.array([[ 0, 1, 0 ], [ 0, 0, -1 ], [ 1, 0, 0 ]], dtype=np.float32)

    x = sensor_manager.cam_transform.location.x
    y = sensor_manager.cam_transform.location.y
    z = sensor_manager.cam_transform.location.z
    pitch = sensor_manager.cam_transform.rotation.pitch
    roll = sensor_manager.cam_transform.rotation.roll
    yaw = sensor_manager.cam_transform.rotation.yaw
    # roll = values['roll']
    # yaw = values['yaw']
    # f = values['f']
    # Cx = values['Cx']
    # Cy = values['Cy']

    # K = np.array([[f, 0, Cx], [0, f, Cy], [0, 0, 1]], dtype=np.float64)

    c_y = np.cos(np.radians(yaw))
    s_y = np.sin(np.radians(yaw))
    c_r = np.cos(np.radians(roll))
    s_r = np.sin(np.radians(roll))
    c_p = np.cos(np.radians(pitch))
    s_p = np.sin(np.radians(pitch))
    matrix = np.identity(4)
    matrix[0, 3] = x
    matrix[1, 3] = y
    matrix[2, 3] = z
    matrix[0, 0] = c_p * c_y
    matrix[0, 1] = c_y * s_p * s_r - s_y * c_r
    matrix[0, 2] = -c_y * s_p * c_r - s_y * s_r
    matrix[1, 0] = s_y * c_p
    matrix[1, 1] = s_y * s_p * s_r + c_y * c_r
    matrix[1, 2] = -s_y * s_p * c_r + c_y * s_r
    matrix[2, 0] = s_p
    matrix[2, 1] = -c_p * s_r
    matrix[2, 2] = c_p * c_r
    matrix = np.linalg.inv(matrix)
    
    P = K @ flip @ matrix[:3, :]

    return P`

`def brute_force_homography_inversion(h_world_to_pixel, width = 600, height = 400) :
    proj_pts = []
    world_pts = []
    while len(proj_pts) < 8:
        x = np.random.randint(-50, 50)
        y = np.random.randint(-50, 50)
        world_pt = np.array([x, y, 1, 1])
        proj = np.dot(h_world_to_pixel, world_pt)
        proj = proj / proj[2]
        if(proj[0]>0 and proj[0]<width and proj[1]>0 and proj[1]<height):
            proj_pts.append(proj)
            world_pts.append(world_pt[:3])
    invH = cv2.findHomography(np.array(proj_pts), np.array(world_pts))
    return invH[0]`

linthmwan, 2023-01-08:
I just found out a solution to alleviate the precision problem. 
Although the inverse of the projection matrix is not accuracy enough, but it's still precise when pixel depth is not far away, e.g. 1.
So we can still can get a precise enough world point when the pixel depth is 1, but we know that this world point is not our desired point, for example, we want a world point it's z location is 1.

But we can form a vector from your camera location and precise world point, then adjust this vector and add the precise world point to get your desired z location!!

Here is the code~
```
def get_camera_world_view(pixel_cor, K_inverse, c2w, desired_z, sensor_location):
    n = pixel_cor.shape[0]
    v = np.concatenate((pixel_cor, np.array([[1.] * n]).T),axis = 1).T 
    F = np.array([[ 0,  1,  0 ], [ 0,  0, -1 ], [ 1,  0,  0 ]], dtype=np.float64)
    F = np.linalg.inv(F)
    to_camera = np.matmul(K_inverse, v)
    to_camera = np.matmul(F, to_camera)
    to_camera = np.concatenate((to_camera, np.array([[1] * n])))
    to_world = np.matmul(c2w, to_camera).T

    vec = to_world[:,:3] - sensor_location.reshape(1,3)
    vec = vec / vec[:,2].reshape(-1,1) * to_world[:,2].reshape(-1,1)

    return to_world[:,:3] - vec
sensor_location = np.array([sensor.get_location().x,sensor.get_location().y,sensor.get_location().z])

world_point = get_camera_world_view(np.array([[1,1.],[1800.,990.]]), K_inverse, c2w, 0, sensor_location)
```
I also verify that this idea is correct by the following function in carla documentation.
```

def get_image_point(loc, K, w2c):
    # Calculate 2D projection of 3D coordinate

    # Format the input coordinate (loc is a carla.Position object)
    point = np.array([loc.x, loc.y, loc.z, 1])
        # transform to camera coordinates
    point_camera = np.dot(w2c, point)

    # New we must change from UE4's coordinate system to an "standard"
    # (x, y ,z) -> (y, -z, x)
    # and we remove the fourth componebonent also
    point_camera = [point_camera[1], -point_camera[2], point_camera[0]]

    # now project 3D->2D using the camera matrix
    point_img = np.dot(K, point_camera)
    # normalize
    point_img[0] /= point_img[2]
    point_img[1] /= point_img[2]

    return point_img[0:2]

point = carla.Location(x=world_point[0,0],y=world_point[0,1],z=world_point[0,2])
get_image_point(point,K,np.array(sensor.get_transform().get_inverse_matrix()))
```
and the output is
array([0.99995575, 1.00001084])
array([1800.0000734 ,  990.00003223]) respectively

I think it is accuracy enough in integer standard.
Enjoy~~

molua, 2023-06-07:
> I just found out a solution to alleviate the precision problem. Although the inverse of the projection matrix is not accuracy enough, but it's still precise when pixel depth is not far away, e.g. 1. So we can still can get a precise enough world point when the pixel depth is 1, but we know that this world point is not our desired point, for example, we want a world point it's z location is 1.
> 
> But we can form a vector from your camera location and precise world point, then adjust this vector and add the precise world point to get your desired z location!!
> 
> Here is the code~
> 
> ```
> def get_camera_world_view(pixel_cor, K_inverse, c2w, desired_z, sensor_location):
>     n = pixel_cor.shape[0]
>     v = np.concatenate((pixel_cor, np.array([[1.] * n]).T),axis = 1).T 
>     F = np.array([[ 0,  1,  0 ], [ 0,  0, -1 ], [ 1,  0,  0 ]], dtype=np.float64)
>     F = np.linalg.inv(F)
>     to_camera = np.matmul(K_inverse, v)
>     to_camera = np.matmul(F, to_camera)
>     to_camera = np.concatenate((to_camera, np.array([[1] * n])))
>     to_world = np.matmul(c2w, to_camera).T
> 
>     vec = to_world[:,:3] - sensor_location.reshape(1,3)
>     vec = vec / vec[:,2].reshape(-1,1) * (to_world[:,2].reshape(-1,1) - desired_z)
> 
>     return to_world[:,:3] - vec
> sensor_location = np.array([sensor.get_location().x,sensor.get_location().y,sensor.get_location().z])
> 
> world_point = get_camera_world_view(np.array([[1,1.],[1800.,990.]]), K_inverse, c2w, 0, sensor_location)
> ```
> 
> I also verify that this idea is correct by the following function in carla documentation.
> 
> ```
> 
> def get_image_point(loc, K, w2c):
>     # Calculate 2D projection of 3D coordinate
> 
>     # Format the input coordinate (loc is a carla.Position object)
>     point = np.array([loc.x, loc.y, loc.z, 1])
>         # transform to camera coordinates
>     point_camera = np.dot(w2c, point)
> 
>     # New we must change from UE4's coordinate system to an "standard"
>     # (x, y ,z) -> (y, -z, x)
>     # and we remove the fourth componebonent also
>     point_camera = [point_camera[1], -point_camera[2], point_camera[0]]
> 
>     # now project 3D->2D using the camera matrix
>     point_img = np.dot(K, point_camera)
>     # normalize
>     point_img[0] /= point_img[2]
>     point_img[1] /= point_img[2]
> 
>     return point_img[0:2]
> 
> point = carla.Location(x=world_point[0,0],y=world_point[0,1],z=world_point[0,2])
> get_image_point(point,K,np.array(sensor.get_transform().get_inverse_matrix()))
> ```
> 
> and the output is array([0.99995575, 1.00001084]) array([1800.0000734 , 990.00003223]) respectively
> 
> I think it is accuracy enough in integer standard. Enjoy~~

may i ask how to get the K_inverse and c2w?

linthmwan, 2023-06-08:
`world_2_camera = np.array(camera.get_transform().get_inverse_matrix())` can get the w2c matrix
```
def build_projection_matrix(w, h, fov):
    focal = w / (2.0 * np.tan(fov * np.pi / 360.0))
    K = np.identity(3)
    K[0, 0] = K[1, 1] = focal
    K[0, 2] = w / 2.0
    K[1, 2] = h / 2.0
    return K

# Get the attributes from the camera
image_w = camera_bp.get_attribute("image_size_x").as_int()
image_h = camera_bp.get_attribute("image_size_y").as_int()
fov = camera_bp.get_attribute("fov").as_float()

# Calculate the camera projection matrix to project from 3D -> 2D
K = build_projection_matrix(image_w, image_h, fov)
```

can get K.

https://carla.readthedocs.io/en/latest/tuto_G_bounding_boxes/
Here is the tutorial for CARLA coordination transform, you can find more detail explanation.
w2c is the matrix to transform world coordinate to camera coordinate.
K is the matrix to transform camera coordinate to image pixel coordinate.