carla-simulator/carla #5568: Unknown error when spawning custom model using try_spawn_actor()

erikgust2, 2022-07-07:
CARLA version: 0.13.9
Platform/OS: Ubuntu 18.04

The problem:
Hello my friends!

I am trying to spawn my own custom static objects through the API similarly to how you do it with cars and pedestrians. However, simply testing it out by running
```
custom_bp = bp_lib.find('static.prop.custom_model')
custom = world.try_spawn_actor(custom_bp, random.choice(spawn_points))
```
doesn't seem to work. It throws the following errors:
```
[2022.07.07-10.46.37:711][646]LogCarla: Spawning actor 'static.prop.custommodel'
[2022.07.07-10.46.37:712][646]LogCarla: Warning: Failed to spawn actor 'static.prop.custommodel'
[2022.07.07-10.46.37:712][646]LogCarla: Error: Actor not Spawned
[2022.07.07-10.46.37:712][646]LogCarlaServer: Responding error: Unknown error while trying to spawn actor
```
Interestingly enough, I can spawn any of the static objects included in the simulation, but not any of the custom ones that I've imported. They can be spawned just fine by manually dragging them from the library into the viewport though, so I am probably doing something very obvious wrong. Any ideas?

For context, I am using a normal .fbx-mesh and added it with the make import-command.

What I expected to happen: the static object spawning at the designated coordinate/spawn point.

AndrewWu1998, 2023-02-25:
Hello, has this problem been solved? I've been stuck in this for a week

lsangreg, 2023-09-22:
Hello @AndrewWu1998 @erikgust2  did you solve the error? I am also stuck!

Blyron, 2026-11-12:
This is about the Unreal Engine 4 `make import` flow for props, which no longer exists. In CARLA 1.0 (Unreal Engine 5.8, branch `ue58-dev`) props are imported as described in [Custom props](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/content_authoring_props.md) and can be shipped in a [content pack](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/content_authoring_packs.md). Closing as obsolete. If it still happens there, please open a new issue with the steps to reproduce and the logs.

closed 2026-11-12 by Blyron