carla-simulator/carla #5569: make launch Error6

VPA2998, 2022-07-07:
I am facing the "make: *** [CarlaUE4Editor] Error 6" for long time please find a best solution for this,
CARLA version: latest
Platform/OS: windows 10
Problem you have experienced: make: *** [CarlaUE4Editor] Error 6
What you expected to happen: make launch

  C:/CARLA/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp(170): error C2440: 'return': cannot convert from 'FTaggedHierarchicalStaticMeshSceneProxy *' to 'FPrimitiveSceneProxy *'
  C:/CARLA/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp(170): note: Types pointed to are unrelated; conversion requires reinterpret_cast, C-style cast or function-style cast
  C:/CARLA/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp(237): error C2039: 'NotDrawTaggedComponents': is not a member of 'FEngineShowFlags'
  C:\UE_4.26\Engine\Source\Runtime\Engine\Classes\Components/BillboardComponent.h(13): note: see declaration of 'FEngineShowFlags'
  C:/CARLA/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp(265): error C2039: 'NotDrawTaggedComponents': is not a member of 'FEngineShowFlags'
  C:\UE_4.26\Engine\Source\Runtime\Engine\Classes\Components/BillboardComponent.h(13): note: see declaration of 'FEngineShowFlags'
  C:/CARLA/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp(291): error C2039: 'NotDrawTaggedComponents': is not a member of 'FEngineShowFlags'
  C:\UE_4.26\Engine\Source\Runtime\Engine\Classes\Components/BillboardComponent.h(13): note: see declaration of 'FEngineShowFlags'
  C:/CARLA/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp(300): error C2614: 'FTaggedHierarchicalStaticMeshSceneProxy': illegal member initialization: 'FHierarchicalStaticMeshSceneProxy' is not a base or member
  C:/CARLA/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp(305): error C2065: 'bVerifyUsedMaterials': undeclared identifier
  C:/CARLA/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp(307): error C2065: 'FLODInfo': undeclared identifier
  C:/CARLA/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp(307): error C2065: 'LODInfo': undeclared identifier
  C:/CARLA/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp(307): error C2143: syntax error: missing ';' before ':'
  C:/CARLA/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp(307): error C2065: 'LODs': undeclared identifier
  C:/CARLA/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp(307): error C2143: syntax error: missing ';' before ')'
  C:/CARLA/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp(308): error C2653: 'FLODInfo': is not a class or namespace name
  C:/CARLA/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp(308): error C2530: 'SectionInfo': references must be initialized
  C:/CARLA/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp(308): error C2143: syntax error: missing ';' before ':'
  C:/CARLA/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp(308): error C2065: 'LODInfo': undeclared identifier
  C:/CARLA/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp(308): error C2143: syntax error: missing ';' before ')'
  C:/CARLA/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp(316): error C2653: 'FHierarchicalStaticMeshSceneProxy': is not a class or namespace name
  C:/CARLA/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp(318): error C2039: 'NotDrawTaggedComponents': is not a member of 'FEngineShowFlags'
  C:\UE_4.26\Engine\Source\Runtime\Engine\Classes\Components/BillboardComponent.h(13): note: see declaration of 'FEngineShowFlags'
  C:/CARLA/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(476): warning C4458: declaration of 'Role' hides class member
  C:\UE_4.26\Engine\Source\Runtime\Engine\Classes\GameFramework/Actor.h(541): note: see declaration of 'AActor::Role'
  C:/CARLA/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(491): warning C4458: declaration of 'Role' hides class member
  C:\UE_4.26\Engine\Source\Runtime\Engine\Classes\GameFramework/Actor.h(541): note: see declaration of 'AActor::Role'
make: *** [CarlaUE4Editor] Error 6

wkgud9090, 2022-07-20:
Have anyone solve this issue?

Blyron, 2025-08-05:
This should be working with latest changes in our Unreal Engine fork

closed 2025-08-05 by Blyron