carla-simulator/carla #5584: roadrunner map singal overlapping

xiaoyang666, 2022-07-14:
[image]
[image]
Hello，
When I draw a map with roadrunner and import the map with make import, the traffic lights at the intersection overlap. I would appreciate if any teacher could tell me why this happens.

xiaoyang666, 2022-07-16:
Hello，everyone

XGodina, 2022-10-10:
Hi, @xiaoyang666 . We used the traffic light head to spawn traffic lights. If you put a traffic light with two heads spawns two traffic lights. My recommendation is to put a traffic light with a head. That should work.