carla-simulator/carla #5586: Unknown error while trying to spawn actor

gistao, 2022-07-14:
CARLA version:0.9.12

Platform/OS:ubuntu18.04

Problem you have experienced:
i got a crash when i spawn a custom actor, the stack:
 `libcarla_client.so!carla::client::detail::Client::SpawnActor(carla::rpc::ActorDescription const&, carla::geom::Transform const&) (未知源:0)

libcarla_client.so!carla::client::detail::Simulator::SpawnActor(carla::client::ActorBlueprint const&, carla::geom::Transform const&, carla::client::Actor*, carla::rpc::AttachmentType, carla::client::GarbageCollectionPolicy) (未知源:0)

libcarla_client.so!carla::client::World::SpawnActor(carla::client::ActorBlueprint const&, carla::geom::Transform const&, carla::client::Actor*, carla::rpc::AttachmentType) (未知源:0)`

What you expected to happen:
spawn a actor when bultin a static prop with the same code

Other information (documentation you consulted, workarounds you tried):
step1: read assets
Import/
├── AvmPkgChair
│   ├── AvmPkgChair.json
│   └── Props
│       └── AvmPrpchair
│           ├── AvmPrpchair.fbm
│           │   ├── padding_bump.jpg
│           │   ├── padding_color.jpg
│           │   ├── pillows_bump.jpg
│           │   ├── pillows_color.jpg
│           │   ├── shell_color.jpg
│           │   └── trim_color.jpg
│           └── AvmPrpchair.fbx

step2: make import, but have warning:
[2022.07.14-12.33.11:382][  0]LogInit: Display: Warning/Error Summary (Unique only)
[2022.07.14-12.33.11:382][  0]LogInit: Display: -----------------------------------
[2022.07.14-12.33.11:382][  0]LogInit: Display: LogConsoleManager: Warning: Setting the console variable 'r.Streaming.PoolSize' with 'SetByScalability' was ignored as it is lower priority than the previous 'SetByProjectSetting'. Value remains '4000'
[2022.07.14-12.33.11:382][  0]LogInit: Display: LogScript: Warning: Script Msg: A null object was passed as a world context object to UEngine::GetWorldFromContextObject().
[2022.07.14-12.33.11:382][  0]LogInit: Display: FBXImport: Warning: No smoothing group information was found in this FBX scene.  Please make sure to enable the 'Export Smoothing Groups' option in the FBX Exporter plug-in before exporting the file.  Even for tools that don't support smoothing groups, the FBX Exporter will generate appropriate smoothing data at export-time so that correct vertex normals can be inferred while importing. 
[2022.07.14-12.33.11:382][  0]LogInit: Display: 
[2022.07.14-12.33.11:382][  0]LogInit: Display: Success - 0 error(s), 7 warning(s)

step3: import asset, success!

step4: find my prop in blueprint lib, but spawn error

lsangreg, 2023-09-22:
Hi @gistao  were you able to solve the error?  I am stuck since a week !

Blyron, 2026-11-12:
This is about the Unreal Engine 4 `make import` flow for props, which no longer exists. In CARLA 1.0 (Unreal Engine 5.8, branch `ue58-dev`) props are imported as described in [Custom props](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/content_authoring_props.md) and can be shipped in a [content pack](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/content_authoring_packs.md). Closing as obsolete. If it still happens there, please open a new issue with the steps to reproduce and the logs.

closed 2026-11-12 by Blyron