carla-simulator/carla #5587: Carla simulator doesn't execute the vehicle control command correctly

MaisJamal, 2022-07-14:
Hello, thanks for creating such a useful simulator.

CARLA version: 0.9.6 (in docker container)
Platform/OS: Ubuntu 20.04
Problem you have experienced: Carla simulator doesn't execute the vehicle control command correctly 
Steps to reproduce:
I run autopilot from Apollo autonomous driving platform, which is connected to Carla by CarlaCyberBridge. I can receive the correct data of sensors in Apollo from Carla, but there seems to be incompatibility in Control command from Apollo to Carla. As you can see in this [video](https://youtu.be/_3WmLRIjRCg), the ego vehicle is not following the correct control.

**_VehicleControlCMD in Apollo:_**
**throttle** (double) target throttle in percentage [0, 100]
**brake** (double) target brake in percentage [0, 100]
**steering_rate** (double) target non-directional steering rate, in percentage of full scale per second [0, 100]
**steering_target** (double) target steering angle, in percentage of full scale [-100, 100]
**parking_brake** (bool) parking brake engage. true: engaged
**gear_location**  GEAR_REVERSE if the ego is in reverse mode

**_VehicleControl in Carla:_**
**throttle** (float) A scalar value to control the vehicle throttle [0.0, 1.0]. Default is 0.0.
**steer** (float) A scalar value to control the vehicle steering [-1.0, 1.0]. Default is 0.0.
**brake** (float) A scalar value to control the vehicle brake [0.0, 1.0]. Default is 0.0.
**hand_brake** (bool) Determines whether hand brake will be used. Default is False.
**reverse** (bool) Determines whether the vehicle will move backwards. Default is False.
**manual_gear_shift** (bool) Determines whether the vehicle will be controlled by changing gears manually. Default is False.
**gear** (int) States which gear is the vehicle running on.

What I do to convert the messages:
vehicle_control = VehicleControl()
vehicle_control.hand_brake = cyber_vehicle_control.parking_brake
vehicle_control.brake = cyber_vehicle_control.brake / 100.0
vehicle_control.steer = cyber_vehicle_control.steering_target / 100.0
vehicle_control.throttle = cyber_vehicle_control.throttle / 100.0
vehicle_control.reverse = cyber_vehicle_control.gear_location == Chassis.GearPosition.GEAR_REVERSE
self.carla_actor.apply_control(vehicle_control)

Can you please guide me on converting the vehicle command to Carla? Is there anything that I'm doing wrong?

Ezharjan, 2024-01-30:
I'm having the similiar issue that the vehicle_control doesn't work at all. 
My code is as follows, under the Carla version 0.9.13:
```python
vehicle_control = vehicle.get_control()
vehicle_control.steer = -0.2 # A scalar value to control the vehicle steering [-1.0, 1.0]. Default is 0.0. 
vehicle.apply_control(vehicle_control) # Not working!!!
```

Blyron, 2026-11-12:
This is a control-mapping problem in the external Apollo / CyberBridge on CARLA 0.9.6, not in the simulator. Closing. If `vehicle.apply_control()` itself misbehaves on a current CARLA, please open a new issue.

closed 2026-11-12 by Blyron