carla-simulator/carla #5593: lidar intensity

kms8527, 2022-07-18:
[image]

Hello.
I visualized lidar scan in rviz. But lidar intensity only changes with distance.
Also i saw the issue below.
Better LiDAR intensity measurements #3826
[image]

Is there any way to visualize like this?

fgoudreault, 2022-07-21:
Not at the moment since the picture you show is an implementation by somebody else not included in Carla.

Paolo08, 2023-11-13:
Any updates on this?

Subhash-Bandi, 2024-02-19:
Hi, I would like to know if any one have update on this.

I need intensity values to be based on material laser is striking than based on just distance.

Paolo08, 2024-03-11:
Hi @Subhash-Bandi,
Check this https://github.com/carla-simulator/carla/issues/3826 as example to improve the intensity measurements of the LiDAR sensor.