carla-simulator/carla #5594: Import left hand drive custom map

StephanieMussi, 2022-07-18:
[image]
CARLA version: 0.9.12
Platform/OS: Ubuntu 18.04
Problem you have experienced: I am trying to import a custom map from RoadRunner R2021b with left-hand driving direction. The map is exported to OpenDrive 1.4 with Driving Side "Left". However, after import the map to Carla, it's shown that the routes are still perceived as right hand driving. Any suggestion on how to make Carla work with left hand driving? Thanks!

Map in RoadRunner (OpenDrive view):
[image]

Map in Carla:
[image]

Export settings:
[image]

StephanieMussi, 2022-08-05:
With reference to issue #2442, I managed to get the route planners generated at expected positions. The direction of spawn points, however, still follows right-hand drive manners. 
Even after rotating every spawn point for 180 degrees, the vehicles spawned with *generate_traffic.py* will try to make U-turn and disappear after a short while.

closed 2022-09-02 by StephanieMussi

JIANGZhengm, 2024-09-11:
@

> With reference to issue #2442, I managed to get the route planners generated at expected positions. The direction of spawn points, however, still follows right-hand drive manners. Even after [...]

hello, have you ever resolved this issue? I am using CARLA 0.9.15, I am looking forward to your reply.

powderly12, 2025-01-28:
@SafeDL I am now facing the same issue and did you find a solution? (also on CARLA 0.9.15) Thank you for any help

Abhilashsabhi, 2025-03-25:
Is there any solution anyone has? I am using Carla 0.9.15.
please help

powderly12, 2025-03-25:
@Abhilashsabhi Hi, I fixed my problem by exporting the road runner file to Carla using the Carla filmbox method. I was modelling a highway scenario, and for some reason, when I specified right-hand driving while exporting the Carla file, it would produce a left-hand driving Carla map after it was imported. I am not sure why, but it allowed left-hand driving in my case. I hope this helps