carla-simulator/carla #5600: How to user carla.map.get_waypoint to get a non-road center point.

yunmo555, 2022-07-19:
Hello, I have a problem about Carla.Waypoint. Thank you in advance.

I expected to happen:
I have a special scenario.  It lets obstacle vehicle drive at the boundary between the two lanes.
I want carla.map.get_waypoint to generate a waypoint of current location.

`
map_wp = None
if not self._generated_waypoint_list:
    map_wp = CarlaDataProvider.get_map().get_waypoint(CarlaDataProvider.get_location(self._actor), project_to_road=False)
else:
    map_wp = CarlaDataProvider.get_map().get_waypoint(self._generated_waypoint_list[-1].location)
while len(self._generated_waypoint_list) < 50:
    map_wps = map_wp.next(2)
    if map_wps:
        self._generated_waypoint_list.append(map_wps[0].transform)
        map_wp = map_wps[0]
    else:
        break
`
`project_to_road=False`  doesn't take effect.
In addition, I try to set trasform for waypoint, but it error `Don't set attribute`.
So, how to get a non-road center waypoint. 

CARLA version: carla-0.9.12
Platform/OS: Ubuntu 20.04

SMaoIM, 2023-02-13:
Hello, have u find a solution now? I have faced the same problem.

yunmo555, 2023-02-13:
> Hello, have u find a solution now? I have faced the same problem.

Yes，I modified it based on scenario runner. 
In simple_vehicle_control.py，you can set a variable to save your offset，then set offset_waypoint when setting new velocity.

`
     direction_norm = self._set_new_velocity(self._offset_waypoint(self._waypoints[0]))
     ......

def _offset_waypoint(self, transform):
        """
        Given a transform (which should be the position of a waypoint), displaces it to the side,
        according to a given offset
        
        Args:
        transform (carla.Transform): Transform to be moved
        
        returns:
        offset_location (carla.Transform): Moved transform
        """
        if self._offset == 0:
            if isinstance(transform, carla.libcarla.Transform):
                offset_location = transform.location
            else:
                offset = transform[1]
                right_vector = transform[0].get_right_vector()
                offset_location = transform[0].location + carla.Location(x=offset*right_vector.x,
                                             y=offset*right_vector.y)
        else:
            if isinstance(transform, carla.libcarla.Transform):
                right_vector = transform.get_right_vector()
                offset_location = transform.location + carla.Location(x=self._offset*right_vector.x,
                                             y=self._offset*right_vector.y)
            else:
                right_vector = transform[0].get_right_vector()
                offset_location = transform[0].location + carla.Location(x=self._offset*right_vector.x,
                                             y=self._offset*right_vector.y)
            return offset_location

`

Blyron, 2026-11-12:
Waypoints are always at the centre of the lane by design. To drive with a lateral offset, use `BasicAgent.set_offset()` (the LocalPlanner `offset`) or the Traffic Manager's `vehicle_lane_offset()`. Closing.

closed 2026-11-12 by Blyron