carla-simulator/carla #5603: Wrong traffic lights definition when manually importing xodr file

jiawei-fan, 2022-07-19:
Hello everyone, I'm trying to construct a co-simulation plat-form between CARLA and SUMO, yet I got some issues when trying to build the CARLA network. Thank you in advance.

CARLA version: 0.9.13
Platform/OS: Windows 10

I followed the instructions at https://carla.readthedocs.io/en/0.9.13/tuto_M_add_map_alternative/#manual-import (since I don't have Roadrunner) and built a simple network with only one cross-road with fbx and xodr file. The fbx file is created with blender, the xodr file is converted from osm file following the instruction here: https://carla.readthedocs.io/en/latest/tuto_G_openstreetmap/#using-openstreetmaps-in-carla. When I generate the routes with the OpenDriveActor in UE, The problem I met was a wrong positioning of traffic lights: 
[image]
As is shown in the figure, there is one traffic light which is not in its proper position. In addition to that, when I run the co-simulation, I observed that the 4 traffic lights are in one single group, which means there's only one traffic light at green state at a time which is not reasonable ( How could I put these 4 traffic lights into 2 different groups?) Could you please help to analyse the reason of these 2 issues? Or tell me how can I modify the attributes of these traffic lights (which seems to be rather hard to me, since these lights are only generated when I hit the play button in UE4) ? 

The fbx , xodr and osm file are attached so that the problem could be reproduced easily.
[single_cross_road.zip](https://github.com/carla-simulator/carla/files/9142806/single_cross_road.zip)
 Thank you very much!

Remark: There's another minor problem, when the fbx and the xodr routes are imported, their position are not compatible and I had to translate the meshes so that the vehicles run properly on the roads. Does anyone have similar issue?

fangchaooo, 2022-07-21:
This is a complex problem. The Reason is Carla converted the OpenDrive map coordinate difference with sumo.

xiaoyang666, 2022-07-21:
> This is a complex problem. The Reason is Carla converted the OpenDrive map coordinate difference with sumo.

@fangchaooo hi, I imported a map into carla by  plugin method and found that the traffic lights overlap, I have not solved this problem. Is there any good way to solve it?
My version is 0.9.13
Thanks!

fangchaooo, 2022-07-28:
If you convert by plugin method, you should first try carla script. Path is `PythonAPI/util/config.py`