carla-simulator/carla #5613: Documentation for mentioning github username and password should be included

dt1729, 2022-07-24:
https://github.com/carla-simulator/carla/blob/8854804f4d7748e14d937ec763a2912823a7e5f5/Util/Docker/Prerequisites.Dockerfile#L5

We have to mention our username and github key here, it is not mentioned in any documentation, should I submit a PR with the same? Or is it something that is already mentioned in some documentation?

Regards.

Bonifatius94, 2022-08-05:
Hey @dt1729 how did you solve the login issue? I'd love to file a constructive PR if I knew how to authenticate.

For non-docker I know how to create this GitHub auth token stuff, but for the ue4-docker tooling, it's really a black box for me :sweat_smile: 

Thanks for your help in advance :wink:

Bonifatius94, 2022-08-05:
I think I figured it out. When prompted, you need to enter the GitHub access token as "password". This should fix it.

dt1729, 2022-09-12:
I did not get any prompt perhaps because it was a university PC and someone with University had a github info entered there. I had to edit the dockerfile and hardcode that information for my account. If it works on a fresh Linux install this issue shall be closed, we'd need someone to test this maybe?

Bonifatius94, 2022-09-12:
> I had to edit the dockerfile and hardcode that information for my account.

I think you're confusing ue4docker with Prerequisites.Dockerfile. You don't need to hardcode the username and password btw. Rather use the --build-arg argument of the *docker build* command to set an ARG in a Dockerfile 😉

I'm working on a new Dockerfile and wanna propose it soon. Stay tuned 😄

dt1729, 2022-09-15:
I can help write a test for it if you want.

Bonifatius94, 2022-09-15:
@dt1729 Here are my Docker tools as PR: #5750. You're welcome to test it if you want :wink:

Blyron, 2026-11-12:
The Unreal Engine 4 `Prerequisites.Dockerfile` no longer exists. The Docker files for CARLA 1.0 (Unreal Engine 5.8, branch `ue58-dev`) in `Util/Docker` do not need GitHub credentials, see [Running CARLA in Docker](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/build_docker.md). Closing.

closed 2026-11-12 by Blyron