carla-simulator/carla #5618: apply texture to car

JeromeFan, 2022-07-26:
I'm using carla version 0.9.13 on ubuntu 18.04. I'm trying to apply my own texture on the car, `vehicle.set_attribute` seems like its only support to change the color of the vehicle, and `world.apply_color_texture_to_object()` seems like its only work on sth like buildings, not car.

Thank you very much!

winterwindwang, 2022-09-04:
> 

Hi, do you solve this issues?

huruihan, 2022-09-17:
Hi, we have the same question

neilsambhu, 2023-02-08:
I have this same issue with not being able to apply a TGA file I modified in GIMP onto a Tesla Model 3.

neilsambhu, 2023-09-18:
I have compiled CARLA from source using Unreal Engine. I exported the SM_Etron file in *.FBX format to Blender, but the exterior color does not exist in the *.FBX file in Blender. My intention was to (1) add the image texture to the *.FBX file in Blender and (2) reimport the file back into Unreal Engine. How can I preserve the exterior color of the *.FBX file?