carla-simulator/carla #5623: Create a new map in CARLA

golnazraja, 2022-07-27:
Hello CARLA team,

I want to add new map in CARLA. My map doesn't have any road, it is just a harbor area which I want to train my vehicles there.
It is written in the document that to add new map, only RoadRunner can be used. I don't have any .xodr file I don't have any road to define. Also, RoadRunner is closed source and I emailed them to get trial version but they are not answering.

Is there anyway to just add 3D model of world like what we do by default in Unreal Engine, without any requirment for .xodr file and road definition?

I hope you can help me with this issue, since it affects my further tasks in CARLA a lot.

golnazraja, 2022-09-30:
I found the answer while ago..
You can basically use baseMap with no object, and add anything you need to it.

closed 2022-09-30 by golnazraja

iasonastsaousis, 2025-01-26:
baseMap you mean the empty level of unreal engine? Or do you need to generate an empty map inside roadrunner and use the .xodr?

golnazraja, 2025-02-02:
> baseMap you mean the empty level of unreal engine? Or do you need to generate an empty map inside roadrunner and use the .xodr?

yes, an empty map in CARLA.

iasonastsaousis, 2025-04-07:
The thing is, that if your custom map doesn't have an .xodr file, you cannot use the world.get_map() methods and everything related to getting/changing the map within the PythonAPI. Is there any bypass for this? Im reading that you can place a "dummy" .xodr file for your map inside the Content/Carla/Maps/OpenDrive folder, without actually using the road network defined. Is this a bad approach?