carla-simulator/carla #5626: Showing strange error

Dhruva678, 2022-07-28:
import glob
import os
import sys
import random
import time
import numpy as np
import cv2

try:
    sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
except IndexError:
    pass

import carla

IM_WIDTH = 640
IM_HEIGHT = 480

def image(image):
    matrix_representational_data = np.array(image.raw_data)
    reshape_of_image = matrix_representational_data.reshape((IM_HEIGHT, IM_WIDTH, 4))
    live_feed_from_camera = reshape_of_image[:, :, :3]
    image = cv2.cvtColor(live_feed_from_camera, cv2.COLOR_BGR2GRAY)
    cv2.imshow("", image)
    cv2.waitKey(1)
    return

def camera(get_blueprint_of_world):
    camera_sensor = get_blueprint_of_world.find('sensor.camera.depth')
    camera_sensor.set_attribute('image_size_x', f'{IM_WIDTH}')
    camera_sensor.set_attribute('image_size_y', f'{IM_HEIGHT}')
    camera_sensor.set_attribute('fov', '70')
    return camera_sensor

actor_list = []
try:

    client = carla.Client('127.0.0.1', 2000)
    client.set_timeout(2.0)
    world = client.get_world()
    get_blueprint_of_world = world.get_blueprint_library()

    car_model = (get_blueprint_of_world.filter(''))
    spawn_point = random.choice(get_blueprint_of_world.filter('vehicle.*.*'))
    dropped_vehicle = world.spawn_actor(car_model, spawn_point)
    dropped_vehicle.set_autopilot(True)

    simulator_camera_location_rotation = carla.Transform(spawn_point.location, spawn_point.rotation)
    simulator_camera_location_rotation.location += spawn_point.get_forward_vector() * 30
    simulator_camera_location_rotation.rotation.yaw += 180
    simulator_camera_view = world.get_spectator()
    simulator_camera_view.set_transform(simulator_camera_location_rotation)

    camera_sensor = camera(get_blueprint_of_world)
    sensor_camera_spawn_point = carla.Transform(carla.Location(x=2.5, z=0.7))
    sensor = world.spawn_actor(camera_sensor, sensor_camera_spawn_point, attach_to=dropped_vehicle)
    sensor.listen(lambda camera_data: image(camera_data))

    actor_list.append(sensor)
    actor_list.append(dropped_vehicle)

    time.sleep(1000)
finally:
    print('destroying actors')
    for actor in actor_list:
        actor.destroy()
    print('done.')

Dhruva678, 2022-07-28:
[image]

Blyron, 2026-11-12:
The error comes from the script: `random.choice(blueprints.filter(...))` returns a blueprint, not a spawn point, and `spawn_actor` needs a blueprint and a `carla.Transform`, for example one from `world.get_map().get_spawn_points()`. Closing.

closed 2026-11-12 by Blyron