carla-simulator/carla #5627: Spawn collision free objects?

MoritzNekolla, 2022-07-28:
Hey,

is it possible to spawn a bike or similar without collision sensors? In other words, the bike cannot be hit by a vehicle or any other object in the environment. Other objects would simply move through the bike instead of colliding with it.

When looking up collision sensors: `"Collision detectors do not have any configurable attribute."` Thus, we can not do anything about it?

Greetings

Pyrestone, 2022-09-01:
bump, this would be very useful for 3d models of sensors which I would like to mount on top of (i.e. colliding with) vehicles.

one would think that Actor.set_simulate_physics(False) would do this, but it doesn't

Blyron, 2024-07-12:
Currently is not a possibility. Sorry for the incovenience