carla-simulator/carla #5630: UE4 crashes when make import customized map

Onepunchman214, 2022-07-29:


cdefg, 2022-09-30:
The same, no solution, my maps are edited by mathworks RoadRunner

XGodina, 2022-10-10:
Hi@Onepunchman214. When you import a map from RoadRunner with two junctions really close never work well. The problem is that share different behaviors of those junctions and then almost always crash it.

SExpert12, 2023-06-30:
Same happened with me. No geometry issues and checked with Opendrive tool in RoadRunner but it got crashed when import in UE Editor.

Blyron, 2026-11-12:
This is about the Unreal Engine 4 `make import` pipeline, which no longer exists. In CARLA 1.0 (Unreal Engine 5.8, branch `ue58-dev`) maps are imported with `Util/Tools/Import.py` or shipped as content packs, see [Custom maps](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/content_authoring_maps.md) and [Content packs](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/content_authoring_packs.md). Closing as obsolete. If it still happens there, please open a new issue with the steps to reproduce and the logs. As noted in the comments, the crash was caused by junctions placed too close together in RoadRunner.

closed 2026-11-12 by Blyron