carla-simulator/carla #5632: Xercesc error installation problem.

saandasa, 2022-07-30:
I'm having problems with Xercesc installing incorrectly in version 0.9.13 on Win11.

`H:\carla\carla>make PythonAPI
-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --all
-[Setup]: Asynchronous jobs:  16
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Install directory:  "H:\carla\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "H:\carla\carla\Build\"
    -[install_zlib]: A zlib installation already exists.
    -[install_zlib]: Delete "H:\carla\carla\Build\zlib-install\" if you want to force a rebuild.
    -[install_zlib]: Exiting...
-[Setup]: Installing libpng...
    -[install_libpng]: [Batch params]: --build-dir "H:\carla\carla\Build\" --zlib-install-dir "H:\carla\carla\Build\zlib-install\"
    -[install_libpng]: A libpng installation already exists.
    -[install_libpng]: Delete "H:\carla\carla\Build\libpng-1.2.37-install\" if you want to force a rebuild.
    -[install_libpng]: Exiting...
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "H:\carla\carla\Build\"
    -[install_rpclib]: A rpclib installation already exists.
    -[install_rpclib]: Delete "H:\carla\carla\Build\rpclib-install\" if you want to force a rebuild.
    -[install_rpclib]: Exiting...
-[Setup]: Installing Google Test...
    -[install_gtest]: [Batch params]: --build-dir "H:\carla\carla\Build\"
    -[install_gtest]: A Google Test installation already exists.
    -[install_gtest]: Delete "H:\carla\carla\Build\gtest-install\" if you want to force a rebuild.
    -[install_gtest]: Exiting...
-[Setup]: Installing "Recast & Detour"...
    -[install_recast]: [Batch params]: --build-dir "H:\carla\carla\Build\"
    -[install_recast]: A "Recast & Detour" installation already exists.
    -[install_recast]: Delete "H:\carla\carla\Build\recast-0b13b0-install\" if you want to force a rebuild.
    -[install_recast]: Exiting...
-[Setup]: Installing Boost...
    -[install_boost]: [Batch params]: --build-dir "H:\carla\carla\Build\" --toolset msvc-14.2 --version 1.72.0 -j 16
    -[install_boost]: A Boost installation already exists.
    -[install_boost]: Delete "H:\carla\carla\Build\boost-1.72.0-install\" if you want to force a rebuild.
    -[install_boost]: Exiting...
已复制         1 个文件。
已复制         1 个文件。
-[Setup]: Installing Xercesc...
    -[install_xercesc]: [Batch params]: --build-dir "H:\carla\carla\Build\"
    -[install_xercesc]: A xerces installation already exists.
    -[install_xercesc]: Delete "H:\carla\carla\Build\xerces-c-3.2.3-install\" if you want to force a rebuild.
    -[install_xercesc]: Exiting...
系统找不到指定的文件。
-[Setup]: Installing Sqlite3
    -[install_sqlite3]: [Batch params]: --build-dir "H:\carla\carla\Build\"
    -[install_sqlite3]: A Sqlite3 installation already exists.
    -[install_sqlite3]: Delete "H:\carla\carla\Build\sqlite3-install" if you want to force a rebuild.
    -[install_sqlite3]: Exiting...
已复制         1 个文件。
-[Setup]: Installing PROJ
    -[install_proj]: [Batch params]: --build-dir "H:\carla\carla\Build\"
    -[install_proj]: A PROJ installation already exists.
    -[install_proj]: Delete "H:\carla\carla\Build\proj-install" if you want to force a rebuild.
    -[install_proj]: Exiting...
已复制         1 个文件。
-[Setup]: Creating "CMakeLists.txt.in"...
-[Setup]:
   ###########
   # SUCCESS #
   ###########

   IMPORTANT!

   All the CARLA library dependences should be installed now.
   (You can remove all "*-src" folders in H:\carla\carla\Build\ directory)

   You only need the ASSET PACK with all the meshes and textures.

[... 92 lines cut ...]
xerces-c_3.lib(ICUTransService.obj) : error LNK2001: 无法解析的外部符号 u_foldCase_58
xerces-c_3.lib(ICUTransService.obj) : error LNK2001: 无法解析的外部符号 UCNV_FROM_U_CALLBACK_STOP_58
xerces-c_3.lib(ICUTransService.obj) : error LNK2001: 无法解析的外部符号 UCNV_FROM_U_CALLBACK_SUBSTITUTE_58
xerces-c_3.lib(ICUTransService.obj) : error LNK2001: 无法解析的外部符号 ucnv_open_58
xerces-c_3.lib(ICUTransService.obj) : error LNK2001: 无法解析的外部符号 ucnv_openU_58
xerces-c_3.lib(ICUTransService.obj) : error LNK2001: 无法解析的外部符号 ucnv_close_58
xerces-c_3.lib(ICUTransService.obj) : error LNK2001: 无法解析的外部符号 ucnv_getMaxCharSize_58
xerces-c_3.lib(ICUTransService.obj) : error LNK2001: 无法解析的外部符号 ucnv_getMinCharSize_58
xerces-c_3.lib(ICUTransService.obj) : error LNK2001: 无法解析的外部符号 ucnv_setFromUCallBack_58
xerces-c_3.lib(ICUTransService.obj) : error LNK2001: 无法解析的外部符号 ucnv_fromUnicode_58
xerces-c_3.lib(ICUTransService.obj) : error LNK2001: 无法解析的外部符号 ucnv_toUnicode_58
xerces-c_3.lib(ICUTransService.obj) : error LNK2001: 无法解析的外部符号 ucnv_fromUChars_58
xerces-c_3.lib(ICUTransService.obj) : error LNK2001: 无法解析的外部符号 ucnv_toUChars_58
xerces-c_3.lib(RangeToken.obj) : error LNK2001: 无法解析的外部符号 uset_openPatternOptions_58
xerces-c_3.lib(RangeToken.obj) : error LNK2001: 无法解析的外部符号 uset_close_58
xerces-c_3.lib(RangeToken.obj) : error LNK2001: 无法解析的外部符号 uset_serialize_58
xerces-c_3.lib(RangeToken.obj) : error LNK2001: 无法解析的外部符号 uset_getSerializedSet_58
xerces-c_3.lib(RangeToken.obj) : error LNK2001: 无法解析的外部符号 uset_setSerializedToOne_58
xerces-c_3.lib(RangeToken.obj) : error LNK2001: 无法解析的外部符号 uset_getSerializedRangeCount_58
xerces-c_3.lib(RangeToken.obj) : error LNK2001: 无法解析的外部符号 uset_getSerializedRange_58
xerces-c_3.lib(XMLUniCharacter.obj) : error LNK2001: 无法解析的外部符号 u_charType_58
build\lib.win-amd64-cpython-38\carla\libcarla.cp38-win_amd64.pyd : fatal error LNK1120: 24 个无法解析的外部命令
error: command 'D:\\Program Files (x86)\\Visual Studio\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\link.exe' failed with exit code 1120

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "H:\carla\carla\PythonAPI\carla\dist"!`

GGILCARVALHO, 2022-10-24:
I have the same error "Failed to find XercesC (missing: XercesC_VERSION)"

LinyeLi60, 2022-10-25:
Same error,  it sames that [https://github.com/carla-simulator/carla/pull/5850](url) change the xercesc from 3.2.3 to 3.2.4

GGILCARVALHO, 2022-10-25:
This error happens with both 3.2.3 and 3.2.4 versions of Xerces-C

GustavoSilvera, 2022-11-06:
This error happens because your Windows machine is having trouble linking the correct unicode (`unicode/uset.h`) methods used in [RangeToken](https://github.com/apache/xerces-c/blob/4bac3c53e44506591c2892cc19f0452be169b401/src/xercesc/util/regx/RangeToken.cpp#L36-L44) and [ICUTransService](https://github.com/apache/xerces-c/blob/b0e7b3c4dc2e3cd3a4edecf61e7c386773f311d1/src/xercesc/util/Transcoders/ICU/ICUTransService.cpp#L33-L38). However, you don't actually need to use them!

If you instead disable the `XERCES_USE_TRANSCODER_ICU` macro and enable the `XERCES_USE_TRANSCODER_WINDOWS` macro (by flipping the default values in [cmake/XercesTranscoderSelection.cmake](https://github.com/apache/xerces-c/blob/4dc7c00d192dcf8f89e1f9f9e3bc5f9a2db52535/cmake/XercesTranscoderSelection.cmake#L97-L101)) then we avoid this error altogether.

shekharsuman3, 2022-12-02:
Platform:
--     Host: Windows-10.0.19042 AMD64
--     Target: Windows-10.0.19042 AMD64
--     CMake: 3.21.3
--     CMake generator: Visual Studio 16 2019
--     CMake build tool: D:/programs/Microsoft visual studio/visual studio/MSBuild/Current/Bin/MSBuild.exe
--     Compiler: MSVC 19.29.30146.0
--
-- Found Proj: D:/carla/Build/proj-install/lib/proj.lib
-- Found ZLIB: D:/carla/PythonAPI/carla/dependencies/lib/zlib.lib (found version "1.2.13")
-- Enabled features: Windows-10.0.19042 AMD64 MSVC 19.29.30146.0 Release Proj
-- Configuring done
-- Generating done
-- Build files have been written to: D:/carla/Build/osm2odr-visualstudio
Microsoft (R) Build Engine version 16.11.2+f32259642 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  osm2odr.vcxproj -> D:\carla\Build\osm2odr-visualstudio\src\Release\osm2odr.lib
  -- Install configuration: "Release"
-[BuildOSM2ODR]: OSM2ODR has been successfully installed in "D:\carla\PythonAPI\carla\dependencies\"
-[BuildPythonAPI]: [Batch params]: --py3
Building Python API for Python 3.
compiling:
  - source/libcarla/libcarla.cpp
running bdist_egg
running egg_info
writing source\carla.egg-info\PKG-INFO
writing dependency_links to source\carla.egg-info\dependency_links.txt
writing top-level names to source\carla.egg-info\top_level.txt
writing manifest file 'source\carla.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
running install_lib
running build_py
running build_ext
building 'carla.libcarla' extension
D:\programs\Microsoft visual studio\visual studio\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Idependencies/include -IC:\Users\u27e45\AppData\Local\Programs\Python\Python37\include -IC:\Users\u27e45\AppData\Local\Programs\Python\Python37\include "-ID:\programs\Microsoft visual studio\visual studio\VC\Tools\MSVC\14.29.30133\ATLMFC\include" "-ID:\programs\Microsoft visual studio\visual studio\VC\Tools\MSVC\14.29.30133\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt" "-ID:\programs\Microsoft visual studio\visual studio\VC\Tools\MSVC\14.29.30133\ATLMFC\include" "-ID:\programs\Microsoft visual studio\visual studio\VC\Tools\MSVC\14.29.30133\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt" /EHsc /Tpsource/libcarla/libcarla.cpp /Fobuild\temp.win-amd64-3.7\Release\source/libcarla/libcarla.obj /experimental:external /external:W0 /external:I dependencies/include/system /DBOOST_ALL_NO_LIB /DBOOST_PYTHON_STATIC_LIB /DBOOST_ERROR_CODE_HEADER_ONLY /D_WIN32_WINNT=0x0600 /DHAVE_SNPRINTF /DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true /MD
libcarla.cpp
D:\programs\Microsoft visual studio\visual studio\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\link.exe /nologo /INCREMENTAL:NO /LTCG /DLL /MANIFEST:EMBED,ID=2 /MANIFESTUAC:NO /LIBPATH:dependencies/lib /LIBPATH:C:\Users\u27e45\AppData\Local\Programs\Python\Python37\libs /LIBPATH:C:\Users\u27e45\AppData\Local\Programs\Python\Python37\PCbuild\amd64 "/LIBPATH:D:\programs\Microsoft visual studio\visual studio\VC\Tools\MSVC\14.29.30133\ATLMFC\lib\x64" "/LIBPATH:D:\programs\Microsoft visual studio\visual studio\VC\Tools\MSVC\14.29.30133\lib\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\lib\um\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.19041.0\ucrt\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.19041.0\um\x64" "/LIBPATH:D:\programs\Microsoft visual studio\visual studio\VC\Tools\MSVC\14.29.30133\ATLMFC\lib\x86" "/LIBPATH:D:\programs\Microsoft visual studio\visual studio\VC\Tools\MSVC\14.29.30133\lib\x86" "/LIBPATH:C:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\lib\um\x86" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.19041.0\ucrt\x86" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.19041.0\um\x86" /EXPORT:PyInit_libcarla build\temp.win-amd64-3.7\Release\source/libcarla/libcarla.obj /OUT:build\lib.win-amd64-3.7\carla\libcarla.cp37-win_amd64.pyd /IMPLIB:build\temp.win-amd64-3.7\Release\source/libcarla\libcarla.cp37-win_amd64.lib shlwapi.lib Advapi32.lib ole32.lib shell32.lib D:\carla\PythonAPI\carla\dependencies/lib\carla_client.lib D:\carla\PythonAPI\carla\dependencies/lib\Detour.lib D:\carla\PythonAPI\carla\dependencies/lib\DetourCrowd.lib D:\carla\PythonAPI\carla\dependencies/lib\libboost_filesystem-vc142-mt-x64-1_72.lib D:\carla\PythonAPI\carla\dependencies/lib\libboost_python37-vc142-mt-x64-1_72.lib D:\carla\PythonAPI\carla\dependencies/lib\osm2odr.lib D:\carla\PythonAPI\carla\dependencies/lib\proj.lib D:\carla\PythonAPI\carla\dependencies/lib\Recast.lib D:\carla\PythonAPI\carla\dependencies/lib\rpc.lib D:\carla\PythonAPI\carla\dependencies/lib\sqlite3.lib D:\carla\PythonAPI\carla\dependencies/lib\xerces-c_3.lib D: [... cut ...]
proj.lib(sqlite3_utils.obj) : error LNK2001: unresolved external symbol sqlite3_vfs_unregister
proj.lib(proj_json_streaming_writer.obj) : error LNK2001: unresolved external symbol sqlite3_vsnprintf
build\lib.win-amd64-3.7\carla\libcarla.cp37-win_amd64.pyd : fatal error LNK1120: 79 unresolved externals
error: command 'D:\\programs\\Microsoft visual studio\\visual studio\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\link.exe' failed with exit status 1120

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "D:\carla\PythonAPI\carla\dist"!

LucasFKobernic, 2023-01-18:
Hi @saandasa did u solve the probelm?

ViperaA60, 2023-02-06:
> 

So the changes I need to make is:

set(XERCES_USE_TRANSCODER_ICU 0) to set(XERCES_USE_TRANSCODER_ICU 1)
set(XERCES_USE_TRANSCODER_ICU 1) under if statement to set(XERCES_USE_TRANSCODER_ICU 0) under if statement
set(XERCES_USE_TRANSCODER_WINDOWS 0) to set(XERCES_USE_TRANSCODER_WINDOWS 1)
set(XERCES_USE_TRANSCODER_WINDOWS 1) under if statement to set(XERCES_USE_TRANSCODER_WINDOWS 0)

Right?

Babybanana1, 2023-04-08:
The issue could be solved by below settings in install_xercesc.bat

cmake .. -G "Visual Studio 16 2019" -A x64^
  -DCMAKE_INSTALL_PREFIX="%XERCESC_INSTALL_DIR:\=/%"^
  -DBUILD_SHARED_LIBS=OFF^
  -Dtranscoder=windows^
  -Dxmlch-type=uint16_t^
  "%BUILD_DIR%%XERCESC_BASENAME%-%XERCESC_VERSION%-source"
if %errorlevel% neq 0 goto error_cmake

molua, 2023-05-13:
> 

that works for me! thanks

molua, 2023-05-13:
> If install fails due to XCERES, change line 54 in "..\carla\Util\InstallersWin\install_xercesc.bat" change [https://downloads.apache.org/xerces/c/3/sources/xerces-c-%XERCESC_VERSION%.zip](https://downloads.apache.org/xerces/c/3/sources/xerces-c-%25XERCESC_VERSION%25.zip) to  https://archive.apache.org/dist/xerces/c/3/sources/xerces-c-%XERCESC_VERSION%.zip

that works for me! thanks

VidaAPI, 2023-06-14:
> If install fails due to XCERES, change line 54 in "..\carla\Util\InstallersWin\install_xercesc.bat" change [...]

Hi, these links are showing: '404 Not Found' , How can i proceed?

Nitro60zeus, 2023-06-14:
Change the Xercesc version to 3.2.4 in the `install_xercesc.bat` file. 

Path: `carla\Util\InstallersWin\install_xercesc.bat`

Else u can download it manually from here:
https://archive.apache.org/dist/xerces/c/3/sources/

VidaAPI, 2023-06-15:
> `N%` to `3.2.4` in the
@Nitro60zeus 
Thank you. I appreciate your answer for this too: I wanna delete the build version that I have (Proceeded till before make Python API). And just quick install 9.13 version. Is deleting the files sufficient enough? Or should I do something else too?

Nitro60zeus, 2023-06-15:
No problem!

What do you mean by quick install? You will be using a pre-built package? If so, then u can run `make clean` and then delete the build folder.

wwwwsea, 2024-11-18:
> The issue could be solved by below settings in install_xercesc.bat
>
> cmake .. -G "Visual Studio 16 2019" -A x64^ -DCMAKE_INSTALL_PREFIX="%XERCESC_INSTALL_DIR:=/%"^ -DBUILD_SHARED_LIBS=OFF^ [...]

tanks.helpful

Blyron, 2026-11-12:
This is an error from the old Make/`.bat` build of CARLA 0.9.x on Unreal Engine 4. That build system is gone: CARLA 1.0 (Unreal Engine 5.8, branch `ue58-dev`) builds with CMake, which downloads and builds the dependencies itself. See [Build on Windows](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/build_windows_ue5.md). Closing as obsolete. If the current build fails for you, please open a new issue with the full log.

closed 2026-11-12 by Blyron