carla-simulator/carla #5639: AttributeError: module 'traci' has no attribute 'sumolib'

Chionanthus, 2022-08-01:
CARLA version: 0.9.13
Platform/OS: Linux

Hi there,

when i use sumo to run a co-simulation example in the `carla-0.9.13/Co-Simulation/Sumo/examples` folder , and it can run with no error. 

But when i use my own .rou.xml file and .sumocfg file to run co-simulation, it said that in `/Co-Simulation/Sumo/sumo_integration/sumo_simulation.py", line 304,` , `module 'traci' has no attribute 'sumolib'

After i change the line 304 `traci.sumolib.net.readNet(net_file)` to `sumolib.net.readNet(net_file) ` and  no error is reported and the system works normally, and i wanna know is there a small bug in `sumo_simulation.py`?

[image]

jy723, 2022-08-01:
You can try to check the  versions of sumo,the new version is different

DFCui, 2022-08-05:
You can change traci.sumolib.net.readNet to sumolib.net.readNet to fix the error.

oyontalas, 2024-04-01:
File "/Carla/Co-Simulation/Sumo/sumo_integration/sumo_simulation.py", line 304, in _get_sumo_net sumo_net = traci.sumolib.net.readNet(net_file) AttributeError: module 'traci' has no attribute 'sumolib'

最终是找到了问题所在：

这是因为sumolib是独立的包，不在traci里面，你需要把carla/Co-Simulation/Sumo/sumo_integration/sumo_simulation.py的304行的这一句代码改成sumo_net = sumolib.net.readNet(net_file)

hotmedal, 2024-06-28:
> You can change traci.sumolib.net.readNet to sumolib.net.readNet to fix the error.

Can confirm this is still the case.

M-Colley, 2026-10-24:
The suggested correction is present in current `ue4-dev` at `401472072315fbd2e2b9b04974dce4c9304f3cda`. [Commit 80af6025b3d6a86a979486d25d39561e9678b573](https://github.com/carla-simulator/carla/commit/80af6025b3d6a86a979486d25d39561e9678b573) changes the call to `sumolib.net.readNet(net_file)`: [current sumo_simulation.py](https://github.com/carla-simulator/carla/blob/401472072315fbd2e2b9b04974dce4c9304f3cda/Co-Simulation/Sumo/sumo_integration/sumo_simulation.py).

An isolated check of the actual whole modules reproduces the AttributeError on 0.9.13 when `traci` has no `sumolib` attribute; the current UE4 module imports the independent `sumolib` reader and resolves the configured net-file path correctly. The reader/dependencies were test doubles, so this checks the reported attribute/import mistake, not a live SUMO/CARLA co-simulation. An unchanged 0.9.13 package still needs the one-line correction or an updated example.

Blyron, 2026-11-12:
Fixed on `ue4-dev` by 80af6025b (`sumolib.net.readNet`). SUMO co-simulation is not part of the Unreal Engine 5 builds. Closing.

closed 2026-11-12 by Blyron