carla-simulator/carla #5641: Doea carla support unreal engine 5?

SalahSoliman, 2022-08-01:


Shreyas-dotcom, 2022-08-03:
Hey Salah, 

No - Carla does not yet support Unreal Engine 5. It was discussed on discord by German Ros. 

He said: 
`I'm not sure of the benefits of UE5 in terms of applications to vehicles
We are planning to transition eventually, aka when it actually works in all platforms`

https://github.com/carla-simulator/carla/issues/5318

Hope this Helps!

SalahSoliman, 2022-08-03:
Hi Shreyas,

Thanks, that’s helpful indeed.
Do you mind sharing the discord server/link?

Best,

Salah

Shreyas-dotcom, 2022-08-03:
> [CAUTION: This email originated from outside of Concordia. Do not click links or open attachments unless you recognize the sender and know the content is safe.]

Hi Salah,

Here is the link to the Carla Discord: https://discord.com/invite/vNVHXfb
[https://cdn.discordapp.com/icons/444206285647380482/128a0cd902ac8e4fa6b58b15a85f6ec7.jpg?size=256]<https://discord.com/invite/vNVHXfb>
Join the carlasim Discord Server!<https://discord.com/invite/vNVHXfb>
Check out the carlasim community on Discord - hang out with 4,324 other members and enjoy free voice and text chat.
discord.com
Here is where I got the link from: https://twitter.com/carlasimulator/status/995944150402043905?lang=en

Hope this Helps!

Best Regards,
Shreyas

SalahSoliman, 2022-08-03:
Thanks, see you there I guess!

Best,

Salah

glopezdiest, 2022-09-27:
Closing this as it has been answered already

closed 2022-09-27 by glopezdiest