carla-simulator/carla #5644: LowLevelFatalError [File:Unknown] [Line: 136]

zhaoyongqi2022, 2022-08-03:
CARLA version: 0.9.13
Platform/OS: Ubuntu 18.04
Problem you have experienced:

I was trying to load OpenDRIVE file standalone with the command "python3 config.py -x opendrive/TownBig.xodr".  I have successfully visualized the example file "TownBid.xodr". However, when I was trying to load my OpenDRIVE file which was purchased from a company, the following errors appeared:

LowLevelFatalError [File:Unknown] [Line: 136] 
Exception thrown: s_position >= 0.0
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=131160 LargeMemoryPoolOffset=196744 
Malloc Size=131160 LargeMemoryPoolOffset=327928 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault

What could be the reason for that? Thanks in advanced!

Delphily, 2022-08-12:
did you solve it yet?
I met this problem too

zhaoyongqi2022, 2022-08-12:
No, the problem is still there. But I guess the reason could be that the OpenDRIVE file is too big too run. Just try to delete a part of it and retry the process.

mrxiaobai-wen, 2022-09-28:
> 

老兄，按照你这个猜想，最终解决问题了吗？

lichengguang, 2024-12-20:
哥们  问题解决了吗？我也遇到了相同的问题

ftgTUGraz, 2024-12-20:
> 哥们 问题解决了吗？我也遇到了相同的问题

可能地图文件太大，删除一些再导入试试

zhaoyongqi2022, 2024-12-20:
> 哥们 问题解决了吗？我也遇到了相同的问题
可能地图文件太大了，删除一部分再试试

Tom6662c, 2025-02-13:
尝试把geforce experience 设置中的自动优化游戏选项关掉，具体可以参考https://www.youtube.com/watch?v=62_ilxOrVYE这个视频第三个解决方法。