carla-simulator/carla #5648: make launch error 1: Util/BuildTools/Linux.mk:137: recipe for target 'setup' failed make: *** [setup] Error 1

anvigo12, 2022-08-05:
Hello,

CARLA version: 0.9.13
Platform/OS: Ubuntu 18.04
Problem you have experienced: make: *** [launch] Error 1
What you expected to happen: launch Carla in server mode or use the Unreal Engine editor
Steps to reproduce: followed documentation till Build Carla step 2
Other information (documentation you consulted, workarounds you tried): 
tried to run without using sudo and getting error:  cp: cannot create regular file '/home/test/carla/PythonAPI/carla/dependencies/lib/libxerces-c.a': Permission denied
when tried to run using sudo getting error: make[1]: Leaving directory '/home/test/carla/Unreal/CarlaUE4'
BuildCarlaUE4.sh: Launching UE4Editor...
Refusing to run with the root privileges.
Util/BuildTools/Linux.mk:7: recipe for target 'launch' failed
make: *** [launch] Error 1

I successfully followed the documentation to run Carla 0.9.13 until the step to compile the server and now getting the following error:

test@test:~/carla$ sudo make launch
[sudo] password for test: 
Setup.sh: llvm-8.0 already installed.
Setup.sh: boost-1.72.0-c8 already installed.
Setup.sh: rpclib-v2.2.1_c5-c8 already installed.
Setup.sh: gtest-1.8.1-c8 already installed.
Setup.sh: recast-0b13b0-c8 already installed.
Setup.sh: Libpng already installed.
Setup.sh: Xerces-c already installed.
Setup.sh: Sqlite already installed.
Setup.sh: PROJ already installed.
Setup.sh: Patchelf already installed.
Setup.sh: CARLA version 0.9.13-1-g8854804f4-dirty.
Setup.sh: Generating CMake configuration files.
Setup.sh: Success!
BuildLibCarla.sh: Building LibCarla "Server.Release" configuration.
ninja: no work to do.
[0/1] Install the project...
-- Install configuration: "Server"
BuildLibCarla.sh: Success!
BuildCarlaUE4.sh: Using Unreal Engine at '/home/test/UnrealEngine_4.26'
BuildCarlaUE4.sh: Build CarlaUE4 project.
make[1]: Entering directory '/home/test/carla/Unreal/CarlaUE4'
bash "/home/test/UnrealEngine_4.26/Engine/Build/BatchFiles/Linux/RunMono.sh" "/home/test/UnrealEngine_4.26/Engine/Binaries/DotNET/UnrealBuildTool.exe" CarlaUE4Editor Linux Development  -project="/home/test/carla/Unreal/CarlaUE4/CarlaUE4.uproject" 

Running Mono...

Fixing inconsistent case in filenames.
Setting up Mono
~/UnrealEngine_4.26/Engine ~/carla/Unreal/CarlaUE4
Using 'git status' to determine working set for adaptive non-unity build (/home/test/UnrealEngine_4.26).
Target is up to date
Total execution time: 3.03 seconds
make[1]: Leaving directory '/home/test/carla/Unreal/CarlaUE4'
BuildCarlaUE4.sh: Launching UE4Editor...
Refusing to run with the root privileges.
Util/BuildTools/Linux.mk:7: recipe for target 'launch' failed
make: *** [launch] Error 1

test@test:~/carla$ make launch
Setup.sh: llvm-8.0 already installed.
Setup.sh: boost-1.72.0-c8 already installed.
Setup.sh: rpclib-v2.2.1_c5-c8 already installed.
Setup.sh: gtest-1.8.1-c8 already installed.
Setup.sh: recast-0b13b0-c8 already installed.
Setup.sh: Libpng already installed.
Setup.sh: Xerces-c already installed.
cp: cannot create regular file '/home/test/carla/PythonAPI/carla/dependencies/lib/libxerces-c.a': Permission denied
Util/BuildTools/Linux.mk:137: recipe for target 'setup' failed
make: *** [setup] Error 1

Please help as I am confused about what to do next.
Thank you in advance!

caodayue, 2022-08-13:
same question,sad...

Dragonking3x, 2022-09-28:
Same problem
[image]
[image]

Ubuntu 20.04.5 LTS
Python 3.8

The solutions of #3547 don't work.
I guess the same error is in #4650 

I followed the steps in https://carla.readthedocs.io/en/0.9.13/build_linux/
UnrealEngine_4.26 is installed, the right branch is used, and in the .bashrc file
```
export UE4_ROOT=~/UnrealEngine_4.26
```
[image]

The Unreal Engine Editor itself works.
```
cd ~/UnrealEngine_4.26/Engine/Binaries/Linux && ./UE4Editor
```

The error happens with 
https://github.com/carla-simulator/carla/tree/0.9.13
https://github.com/carla-simulator/carla

Dragonking3x, 2022-10-14:
For other users that found this Issue.

Try
`sudo make launch`

You have to use "make launch" after the boots lines, because you will get a `UE4_ROOT is not defined` later.

For me, it worked.

imsrinin, 2022-11-05:
Is anyone able to solve this issue?

I am installing Carla in Ubuntu 20.04 using UE 4.26 (python 3.9)

when I'm trying to compile the server using the command 'make launch' I get the following error:

make[1]: *** [Makefile:315: CarlaUE4Editor] Error 6
make[1]: Leaving directory '/home/snaray23/carla/Unreal/CarlaUE4'
make: *** [Util/BuildTools/Linux.mk:7: launch] Error 2

can anyone help me out?

Dragonking3x, 2022-11-05:
Python 3.9 doesn't work.
You have to use 3.7 or 3.8

"Make launch" is using python3 to do stuff.
If you want to change the python that is used, it is better to set it in the setup file.
[Util](https://github.com/carla-simulator/carla/tree/master/Util)/[BuildTools](https://github.com/carla-simulator/carla/tree/master/Util/BuildTools)/Setup.sh

If thay didn't work, test if Unreal is installed correctly. 
- `./Setup.sh`
- `./GenerateProjectFiles.sh`
- `make`

If it still don't work, try to add this like to your .bashrc
`export CPLUS_INCLUDE_PATH="$CPLUS_INCLUDE_PATH:/usr/bin/python3.8"`
If you may have to change the path to the right python version.

imsrinin, 2022-11-09:
I got the error solved. I was using UEs own 4.26.1 version but we need to use the Carla variant of UE. After changing my UE, giving a clean, PythonAPI and launch solved the issue. I was still using python 3.9 while solving the error.

Wanchor0729, 2023-07-21:
same question!!!!   I installed 0.9.14 for success,when I changed 0.9.13,I get the error.

UPTOLIMIT, 2023-10-18:
I am developing Carla using Ubuntu22.04 and Python3.10. After some adjustments, the build finally **succeeded.** 
[image]

My experience is as follows:
If an error is reported:

> Util/BuildTools/Linux.mk:101: recipe for target ‘setup’ failed
> make: *** [setup] Error 128

Modify line 121 of **/home/xiaomi/carla/Util/BuildTools/setup.sh**
**BOOST_VERSION=1.80.0=>BOOST_VERSION=1.81.0**
If an error is reported:

> CMake Error at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
>    Failed to find XercesC (missing: XercesC_VERSION)
> Call Stack (most recent call first):
>    /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
>    /usr/share/cmake-3.22/Modules/FindXercesC.cmake:110 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
>    CMakeLists.txt:71 (find_package)
> 
> -- Configuring incomplete, errors occurred!
> See also "/home/xiaomi/carla/Build/libosm2dr-build/CMakeFiles/CMakeOutput.log".
> make: *** [Util/BuildTools/Linux.mk:152: osm2odr] Error 1

Modify: **/home/xiaomi/carla/Util/BuildTools/setup.sh line 432
XERCESC_VERSION=3.2.3=>XERCESC_VERSION=3.2.4**
**/home/xiaomi/carla/Util/BuildTools/BuildOSM2ODR.sh lines 108 and 109
xerces-c-3.2.3-install=>xerces-c-3.2.4-install**

yuan-mingxi, 2024-02-21:
ok i finally success but idnot know if the solution can help you guys or not

see link“https://link.zhihu.com/?target=https%3A//ftp.cixug.es/apache/xerces/c/3/sources/”for newest version
修改一：~/carla/Util/BuildTools/Setup.sh      428lins change to  XERCESC_VERSION=3.2.5
修改二：~/Carla/Util/BuildTools/BuildOSM2ODR.sh 100lines change to below
      -G "Eclipse CDT4 - Ninja" \
      -DCMAKE_INSTALL_PREFIX=${LIBCARLA_INSTALL_CLIENT_FOLDER} \
       -DPROJ_INCLUDE_DIR=${CARLA_BUILD_FOLDER}/proj-install/include \
      -DPROJ_LIBRARY=${CARLA_BUILD_FOLDER}/proj-install/lib/libproj.a \
      -DXercesC_INCLUDE_DIR=${CARLA_BUILD_FOLDER}/xerces-c-3.2.5-install/include \
      -DXercesC_LIBRARY=${CARLA_BUILD_FOLDER}/xerces-c-3.2.5-install/lib/libxerces-c.a \ 
  ninja osm2odr
  ninja install

and  repeat 

> make PythonAPI

yuan-mingxi, 2024-02-22:
For other users that found this Issue.
> 
> Try `sudo make launch`
> 
> You have to use "make launch" after the boots lines, because you will get a `UE4_ROOT is not defined` later.
> 
> For me, it worked.
but i came across the error like this:
BuildCarlaUE4.sh: Launching UE4Editor...
Refusing to run with the root privileges.
Util/BuildTools/Linux.mk:7: recipe for target 'launch' failed
make: *** [launch] Error 1

PatrickPromitzer, 2024-02-22:
I should explain the solution a bit, to make it easier to understand (I switched accounts)

Carla is compiling one package after each other.
If you run the "make launch" command again, it will look for installed packages (installed in the Carla folder), and skip the existing one.

One of the packages had a problem with the "make launch" permission or other problems with non-root permissions.

For that reason, if you use "sudo make launch", it will give the package compiler root permissions, but not every package wants root permissions.

boots want the "UE4_ROOT" set, which is not set on the root account but on your user account.
For that reason you have to use "make launch" for boots

For example:
packages
1: non-root
2: non-root
3: root
4: non-root
5: non-root

With "make launch" it works for packages 1 and 2 and then gives an error.
If you use "sudo make launch" after that, it will skip packages 1 and 2, and work on package 3.
Package 4 needs a non-root again, it will give an error with "sudo make launch", but you can use "make launch" because you are done with package 3.

I edited my Setup.sh file after that and don't have to go in that direction again.
If you don't want to do that, just switch between "make launch" and "sudo make launch" until it works.
If it doesn't work, it could have another problem or the solution is outdated.

These are the changes I made to the setup.sh file until now (0.9.13).
carla/Util/BuildTools/Setup.sh
```
# line 15 (old)
VERSION_LIST=3
# line 15 (new)
VERSION_LIST=3.8

# line 133 (old)
wget "https://boostorg.jfrog.io/artifactory/main/release/${BOOST_VERSION}/source/${BOOST_PACKAGE_BASENAME}.tar.gz" || true
# line 133 (new)
wget "https://carla-releases.s3.eu-west-3.amazonaws.com/Backup/${BOOST_PACKAGE_BASENAME}.tar.gz" || true

# line 432 (old)
XERCESC_REPO=https://ftp.cixug.es/apache//xerces/c/3/sources/xerces-c-${XERCESC_VERSION}.tar.gz
# line 432 (new)
XERCESC_REPO=https://archive.apache.org/dist/xerces/c/3/sources/xerces-c-${XERCESC_VERSION}.tar.gz
```

I made some changes in the BuildUtilsDocker.sh file, too
carla/Util/BuildTools/BuildUtilsDocker.sh

```
# line 16 (old)
BXSDK_URL=https://www.autodesk.com/content/dam/autodesk/www/adn/fbx/2020-0-1/${LIB_NAME}.tar.gz
# line 16 (new)
FBXSDK_URL=https://damassets.autodesk.net/content/dam/autodesk/www/adn/fbx/2020-0-1/${LIB_NAME}.tar.gz

# line 20 (old)
wget -c "${FBXSDK_URL}" -P "${CARLA_DOCKER_UTILS_FOLDER}"
# line 20 (new)
wget -c "${FBXSDK_URL}" -P "${CARLA_DOCKER_UTILS_FOLDER}" --user-agent="Mozilla"
```

yuan-mingxi, 2024-02-22:
邮件已收到，谢谢！（来自QQ邮箱自动回复邮件）

yuan-mingxi, 2024-02-22:
> thank you very much for your reply. As a beginner using the Carla project, many solving methods are still in the exploratory stage. Your reply has taught me a lot. Thank you! i tried 1.use"make launch",get error boost_1_72_0.tar.gz: Insufficient permissions Unable to write to 'boost_1_72_0.tar.gz' (success). Setup.sh: Extracting boost for Python 3. tar (child): boost_1_72_0.tar.gz: cannot open: No such file or directory tar (child): Error is not recoverable: exiting now tar: Child returned status 2 tar: Error is not recoverable: exiting now Util/BuildTools/Linux.mk:137: recipe for target 'setup' failed make: *** [setup] Error 2 2.use"sudo make launch",get error BuildCarlaUE4.sh: Launching UE4Editor... Refusing to run with the root privileges. Util/BuildTools/Linux.mk:7: recipe for target 'launch' failed make: *** [launch] Error 1 so neither can complete launch work,sad(;--;)

so now iam thinking maybe reinstall my UE4.26

PatrickPromitzer, 2024-02-23:
I forgot to write down, that you have to set the permissions after the "sudo make launch" command because the files were created with root.

It is good that it works now

ARRETdaniel, 2024-10-17:
The error you encountered, specifically "Error 137" and the "Killed" message, typically indicates that the system ran out of memory (OOM, or Out Of Memory). 
Adding more swap space can help if your system runs out of RAM. You can create a swap file to increase the virtual memory.

```
# Check your current swap space
sudo swapon --show

# Create a swap file (e.g., 8GB)
sudo fallocate -l 8G /swapfile

# Set correct permissions
sudo chmod 600 /swapfile

# Set up the swap area
sudo mkswap /swapfile

# Enable the swap file
sudo swapon /swapfile

# Verify if the swap is active
sudo swapon --show

```

yuan-mingxi, 2024-10-17:
your email has been received, thank you!邮件已收到，谢谢！

George-Mats, 2025-09-18:
make sure that you have run `export UE4_ROOT=~/UnrealEngine_4.26` in your shell before `make PythonAPI` etc

I had the same problem and seems to work now

Blyron, 2026-11-12:
This is an error from the old Make/`Setup.sh` build of CARLA 0.9.x on Unreal Engine 4. That build system is gone: CARLA 1.0 (Unreal Engine 5.8, branch `ue58-dev`) builds with CMake and uses the Unreal Engine toolchain. See [Build on Linux](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/build_linux_ue5.md). Closing as obsolete. If the current build fails for you, please open a new issue with the full log.

closed 2026-11-12 by Blyron