carla-simulator/carla #5649: Make PythonAPI error: CMakeLists.txt can't found

caodayue, 2022-08-05:
I got a error:

CMake Error: The source directory "E:/carla/Build/rpclib-src" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.

    -[install_rpclib]: [CMAKE ERROR] An error ocurred while executing the cmake.
    -[install_rpclib]: [CMAKE ERROR] Possible causes:
    -[install_rpclib]:                - Make sure "CMake" is installed.
    -[install_rpclib]:                - Make sure it is available on your Windows "path".
    -[install_rpclib]:                - CMake 3.9.0 or higher is required.
    -[install_rpclib]: Exiting with error...

-[Setup]:
   Ok, and error ocurred, don't panic!
   We have different platforms where you can find some help :)

   - Make sure you have read the documentation:
       http://carla.readthedocs.io/en/latest/how_to_build_on_windows/

   - If the problem persists, you can ask on our Github's "Building on Windows" issue:
       https://github.com/carla-simulator/carla/issues/21

   - Or just use our Discord channel!
       We'll be glad to help you there :)
       https://discord.gg/42KJdRj
make: *** [setup] 错误 1

I try:
1.download zlib1.2.11 
2.modify E:\carla\Util\InstallersWin\install_zlib.bat
but failed.

Who can help me?

caodayue, 2022-08-08:
anybody?

jchen106, 2023-01-06:
how do you solve this problem? i encounter this

tonmoay, 2023-01-16:
I worked around the issue for two days. It turns out as a simple 'Git' issue. 'install_rpclib.bat' uses 'Git' to pull the rpclib repo.

`echo git clone -b "%RPC_VERSION%" https://github.com/carla-simulator/rpclib.git "%RPC_SRC_DIR:~0,-1%"
 call git clone -b "%RPC_VERSION%" https://github.com/carla-simulator/rpclib.git "%RPC_SRC_DIR:~0,-1%"`

CharlieKingOutwood, 2023-02-21:
Deleting "E:/carla/Build/rpclib-src" and making again worked for me.

omitheaussie, 2023-10-12:
Hope this helps someone. I had the same issue with Carla 0.9.13 when I run make PythonAPI. I found out that in the install_zlib.bat line 51 is set ZLIB_VERSION=1.2.13
When the batch file tries to download, it doesnt find the file at www.zlib.net/zlib1.2.13.zip. I found (I think) the reason is the file doesnt exist. The name is different. At https://github.com/madler/zlib/releases/tag/v1.2.13 the file name is zlib1213.zip
So I downloaded that file from github and manually renamed it to zlib1.2.13.zip and saved it in carla/build. Then I ran make PythonAPI. Voila.

Blyron, 2026-11-12:
This is an error from the old Make/`.bat` build of CARLA 0.9.x on Unreal Engine 4. That build system is gone: CARLA 1.0 (Unreal Engine 5.8, branch `ue58-dev`) builds with CMake, which downloads and builds the dependencies itself. See [Build on Windows](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/build_windows_ue5.md). Closing as obsolete. If the current build fails for you, please open a new issue with the full log.

closed 2026-11-12 by Blyron