carla-simulator/carla #5651: More Intelligent Update.sh Script

Bonifatius94, 2022-08-05:
Hi guys,

Your current Update.sh script copies files like crazy. Idk if it's intended, but apparently you're killing disks with it ^^

It's easily possible to pipe the tar.gz download stream of *wget* into the unzipping program *tar*. That way the tar.gz file would never hit the disk because it's streamed via RAM. This allows to download and unzip at the same time which means huge time savings for e.g. the Docker build, etc.

I'm working on a better script and let you know once it's finished.

Best wishes,
Marco

Blyron, 2024-07-12:
Update files will be deprectaed in UE5, if you still want to contribute to UE4 branch we will be pleased

closed 2024-07-12 by Blyron

Bonifatius94, 2024-07-13:
@Blyron Why did it take you 2 years to write this simple reply? You could have told me a lot earlier xD