carla-simulator/carla #5652: ./boost/python/detail/wrap_python.hpp:57:11: fatal error: 'pyconfig.h' file not found

Merealtea, 2022-08-06:
CARLA version: 0.9.13
Platform/OS: Ubuntu:20.04/18.04 
                                    python3.7

Problem you have experienced:
     [image]
    ./boost/python/detail/wrap_python.hpp:57:11: fatal error: 'pyconfig.h' file not found
    I met the problem when I typed the make PythonAPI ARGS=python3.7. I tried in both Ubuntu20.04 and Ubuntu18.04, and they both showed me this problem. 

Steps to reproduce:
      I totally follow the official tutorial here https://carla.readthedocs.io/en/latest/build_linux/. And I successfully built up the UnrealEngin_4.26. And I also successfully finished ./Update.sh. I tried both make PythonAPI and PythonAPI ARGS=python3.7

Merealtea, 2022-08-06:
I noticed that there is no  -I"/home/luban/anaconda3/envs/Carla/include/python3.7" in my environment, but  -I"/home/luban/anaconda3/envs/Carla/include/python3.7m", I guess this caused this problem, but I don`t know how to solve this problem yet.
[image]

Merealtea, 2022-08-06:
Well, I fixed it by 
> ln -s path/to/include/python{version}m path/to/include/python{version}. 
Only add a soft link can avoid this problem, It`s not a big deal. Thanks, I had successfully installed carla right now

closed 2022-08-06 by Merealtea

leishi23, 2022-11-15:
> Well, I fixed it by
> 
> > ln -s path/to/include/python{version}m path/to/include/python{version}.
> > Only add a soft link can avoid this problem, It`s not a big deal. Thanks, I had successfully installed carla right now

Same issue, and solved by the soft link. Thanks!

2019211753, 2023-09-13:
> Well, I fixed it by
> 
> > ln -s path/to/include/python{version}m path/to/include/python{version}.
> > Only add a soft link can avoid this problem, It`s not a big deal. Thanks, I had successfully installed carla right now

also meet this issue, and solved it! thanks a lot.  i installed python interpreter with multithreading，then the pythonx.x'name turned into pythonx.xm

zhuabaguai, 2025-01-24:
great work