carla-simulator/carla #5658: python3 generate_traffic.py , vehicles are not generated.

caodayue, 2022-08-10:
I have built the carla environment,  things all go fine. However, i can't run all .py controller, what's the problem?
 
no errors.
[image]

[image]

carla 0.9.13
window10
python 3.9.4

ccqil, 2022-08-12:
change your python 3.9 to 3.8 or 3.7.
This problem happens when you try to spawn walkers. Actually, if you want only generater vehicles, you can try`python3 generate_traffic.py -w 0`, it will work if you use python 3.9.
If you want to spawn walkers with python 3.9, you should change
https://github.com/carla-simulator/carla/blob/8854804f4d7748e14d937ec763a2912823a7e5f5/PythonAPI/carla/source/libcarla/World.cpp#L305
to
` .def("get_random_location_from_navigation", CALL_RETURNING_OPTIONAL(cc::World, GetRandomLocationFromNavigation)) `

Btw, if you use python 3.9, you will also find `python config.py -l` print nothing.
Delete
https://github.com/carla-simulator/carla/blob/8854804f4d7748e14d937ec763a2912823a7e5f5/PythonAPI/carla/source/libcarla/Client.cpp#L25
will solve this problem.

Blyron, 2025-08-05:
Thanks for yor help. This issue should be fixed in latest versions of CARLA