carla-simulator/carla #5662: Error with C compiler

mspring6, 2022-08-10:
Below is the output from the x64 Native Tools Command Prompt: 

C:\carla>"C:\Program Files (x86)\GnuWin32\bin\make.exe" PythonAPI
-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --all
-[Setup]: Asynchronous jobs:  12
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Install directory:  "C:\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "C:\carla\Build\"
    -[install_zlib]: Not downloading zlib because already exists the folder "C:\carla\Build\zlib-source\".
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is MSVC 19.29.30146.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - broken
CMake Error at C:/Program Files/CMake/share/cmake-3.24/Modules/CMakeTestCCompiler.cmake:69 (message):
  The C compiler

    "C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: C:/carla/Build/zlib-source/build/CMakeFiles/CMakeTmp

    Run Build Command(s):nmake -f Makefile /nologo cmTC_9e161\fast &&   C:\Program Files (x86)\GnuWin32\bin\make.exe  -f CMakeFiles\cmTC_9e161.dir\build.make /nologo -L                  CMakeFiles\cmTC_9e161.dir\build
    'C:\Program' is not recognized as an internal or external command,
    operable program or batch file.
    NMAKE : fatal error U1077: 'C:\Program' : return code '0x1'
    Stop.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:4 (project)

-- Configuring incomplete, errors occurred!
See also "C:/carla/Build/zlib-source/build/CMakeFiles/CMakeOutput.log".
See also "C:/carla/Build/zlib-source/build/CMakeFiles/CMakeError.log".

    -[install_zlib]: [CMAKE ERROR] An error ocurred while executing cmake command.
    -[install_zlib]: [CMAKE ERROR] Possible causes:
    -[install_zlib]:                - Make sure "CMake" is installed.
    -[install_zlib]:                - Make sure it is available on your Windows "path".
    -[install_zlib]:                - Make sure you have cmake 3.12.4 or higher installed.
    -[install_zlib]: Exiting with error...

-[Setup]:
   Ok, and error ocurred, don't panic!
   We have different platforms where you can find some help :)

   - Make sure you have read the documentation:
       http://carla.readthedocs.io/en/latest/how_to_build_on_windows/

   - If the problem persists, you can ask on our Github's "Building on Windows" issue:
       https://github.com/carla-simulator/carla/issues/21

   - Or just use our Discord channel!
       We'll be glad to help you there :)
       https://discord.gg/42KJdRj
make: *** [setup] Error 1

ronyshaji, 2022-09-07:
@mspring6 Did you solve the problem ? I have the same problem.

mspring6, 2022-09-07:
@ronyshaji I did not solve the problem, I just found a work around. I was able to build the carla solution for unreal engine by doing the "make launch" command and that successfully launches unreal engine with carla running. I then just downloaded a pre compiled binary version carla just to get the PythonAPI. Then with the from source version of Unreal Engine server running you should be able to run the PythonAPI client and connect to the server.

ronyshaji, 2022-09-07:
@mspring6 Where did you find the pre compiled version of carla. Any tutorials avalable ?

mspring6, 2022-09-07:
@ronyshaji just go to their website and click on get carla

ronyshaji, 2022-09-08:
@mspring6 How did the make launch work for you ? I mean did you done anything special, when I use the Make Launch, it starts just like MAKE PythonAPI and fails with the same error as above.

bm016662, 2022-09-09:
@mspring6 Thanks for your help. Finally I found a solution to the error and able to launch the UE4 editor.

ZQlby, 2022-09-19:
BEWARE
InstallCleanup.exe now is in here 
"C:\Program Files (x86)\Microsoft Visual Studio\Installer"
instead of 
"To completely clean Visual Studio from the computer, go to Program Files (x86)\Microsoft Visual Studio\Installer\resources\app\layout and run .\InstallCleanup.exe -full"

ronyshaji, 2022-11-09:
@mspring6 
Hallo
I was able to solve the error. In my case the installation directory of gnuwin make had spaces (C:\Program Files\GnuWin32\bin) and i uninstalled from that folder and installed directly to the c drive and added the path to the PATH variable. Reboot the pc and run MakePythonAPI...it works (remember to delete the build folder from the carla root as it was created when you run the makepythonAPI first). May be helpful for some

closed 2024-05-17 by MarcelPiNacy-CVC