carla-simulator/carla #5666: carla docker  ---->Password

KangChou, 2022-08-14:
`carla@ovo:~$ ./CarlaUE4.sh -opengl
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
sh: 1: xdg-user-dir: not found
error: XDG_RUNTIME_DIR not set in the environment.
error: XDG_RUNTIME_DIR not set in the environment.
error: XDG_RUNTIME_DIR not set in the environment.
carla@ovo:~$ su
Password: 

[4]+  Stopped                 su
carla@ovo:~$ su
Password: 
su: Authentication failure
carla@ovo:~$ su
Password: 
su: Authentication failure
`

KangChou, 2022-08-14:
carla@ovo:~$ ./CarlaUE4.sh -opengl
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
sh: 1: xdg-user-dir: not found
error: XDG_RUNTIME_DIR not set in the environment.
error: XDG_RUNTIME_DIR not set in the environment.
error: XDG_RUNTIME_DIR not set in the environment.
carla@ovo:~$ su
Password: 

[4]+  Stopped                 su
carla@ovo:~$ su
Password: 
su: Authentication failure
carla@ovo:~$ su
Password: 
su: Authentication failure

lshmouse, 2022-09-23:
@KangChou Open another terminal and docker exec as root, and you can execute the commands which need root privileges.

```
docker exec -it --user root sleepy_lederberg /bin/bash
```

Blyron, 2026-11-12:
The `carla` user in the container has no root password. Open a root shell with `docker exec -u root -it <container> bash` or start the container with `--user root`. Closing.

closed 2026-11-12 by Blyron