carla-simulator/carla #5674: Custom vehicle being continuously and infinitely spawned

eslaaam3, 2022-08-16:
CARLA version: 0.9.13
Platform/OS: Ubuntu 20.04
Problem you have experienced: Custom vehicle is being continuously spawned 
What you expected to happen: Custom vehicle spawned properly and only once
Steps to reproduce: run `python3 manual_control.py --filter <vehicle>`
Other information (documentation you consulted, workarounds you tried):

- [This answer](https://github.com/carla-simulator/carla/issues/2925#issuecomment-651749779) on this [issue](https://github.com/carla-simulator/carla/issues/2925) #2925 for rigging the vehicle and creating its blueprint.
- [This issue](https://github.com/carla-simulator/carla/issues/4784) #4784 because it has a similar error to the one I had, but unfortunately it didn't help.
- #3789

I have created a custom vehicle in Blender and used the above mentioned resources to create the blueprint in Carla.
But when I try to use `manual_control.py` to spawn the vehicle, it keeps showing the errors shown in pictures below and keeps spawning the vehicle continuously and infinitely. I think every time the spawning fails, a new vehicle is spawned!
I tried running `python3 manual_control.py --filter a2` and the Audi A2 works perfectly.
I also tried modifying Audi A2 in VehicleFactory to add my vehicle's blueprint and ran `python3 manual_control.py --filter a2` again, but the same issue persists.

[image]

[image]

[image]

[image]

gsmario, 2024-05-17:
Hello, sorry for the long time to answer.
Did you solve your problem? If you still need help let me know.

gsmario, 2024-06-07:
I'm closing this issue, feel free to reopen it if you need more help.
Thanks.

closed 2024-06-07 by gsmario