carla-simulator/carla #5675: zLib version

kclarke-ara, 2022-08-16:
CARLA version: latest & 0.9.13
Platform/OS: Windows
Problem you have experienced: Installation Instruction "make PythonAPI" failing
Error returned:
 -[install_zlib]: Retrieving zlib.
Exception calling "DownloadFile" with "2" argument(s): "The remote server returned an error: (404) Not Found."
At line:1 char:1
+ (New-Object System.Net.WebClient).DownloadFile('http://www.zlib.net/z ...
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: (:) [], MethodInvocationException
    + FullyQualifiedErrorId : WebException

What you expected to happen: "make PythonAPI" does not fail
Steps to reproduce: Follow installation documentatation for Windows. At step "make PythonAPI", an error occurs.
Other information (documentation you consulted, workarounds you tried): Updating "instal_zlib.bat" to use version 1.2.12 instead of 1.2.11(updated on March 2022) fixes this issue.

blankdean, 2022-08-21:
I am experiencing the same exact error. Updating "instal_zlib.bat" to use version 1.2.12 fixed the issue for me as well! Thank you

inv-ashikali, 2022-08-28:
> 

can you please help with the update procedure

kclarke-ara, 2022-08-29:
Open the following file from inside Carla: \Util\InstallersWin\install_zlib.bat

Change the following (currently line 51) from:
`set ZLIB_VERSION=1.2.11`

to the following:
`set ZLIB_VERSION=1.2.12`

yzJean, 2022-09-16:
It worked! Many thanks!

ZiadAmmar, 2022-09-20:
Thanks so much. This minor modification has resolved my issue.

AndreasUrlberger, 2022-10-26:
> Open the following file from inside Carla: \Util\InstallersWin\install_zlib.bat
> 
> Change the following (currently line 51) from: `set ZLIB_VERSION=1.2.11`
> 
> to the following: `set ZLIB_VERSION=1.2.12`

Setting it to version 1.2.12 does not work anymore, the current version is 1.2.13. 
The most recent version can be found at http://www.zlib.net/

Ezharjan, 2022-10-29:
> 

This worked fine for me. Maybe someday later this will still not work, then the way to make it work might only be changing the last number of the zlib version in the file. 
But I wonder why zlib removes the old versions to trigger such kind of issues?

closed 2024-05-10 by Blyron