carla-simulator/carla #5678: Fisheye Segmentation looks weird in carla

FrankZTH, 2022-08-17:


arukan404, 2022-08-24:
@FrankZTH I am trying annaornatskaya's fisheye commits to create a fisheye sensor, but it turns out it uses a modified CubemapHelpers class called CubemapHelpersFisheye from UnrealEngine. Any idea how to resolve those errors?

Mnzs1701, 2022-09-02:
@arukan404 have you applied the patch for Unreal Engine for the fisheye lens as mentioned [here](https://github.com/carla-simulator/carla/pull/3755), I have been trying to install the same [patch](https://github.com/carla-simulator/carla/files/5739990/UE4_patch_fisheye-sensor.zip) but it seems to be compatible with Unreal Engine 4.24 and not Unreal Engine 4.26
@FrankZTH could clarify?

arukan404, 2022-09-02:
Hi @Mnzs1701, Thanks for sharing the patch, I guess this is what I have been looking for. Let me try to integrate this and see if it works. Currently, I am doing equirectangular to fisheye projection on the python client side. It works but has a lot of latency... Hopefully, this patch should make it work

arukan404, 2022-09-05:
I got the patch working but now the rendering looks like this
[image]

FrankZTH, 2022-10-04:
@Mnzs1701 You are right, it seems not to be compatible with Unreal Engine 4.26 but I remembered that after del some not support 4.26 codes. It can run with 4.26 too.

closed 2022-11-03 by FrankZTH

rreddy78, 2023-01-25:
> 
Hello @FrankZTH : Could you please post that patch which works with 4.26. Would be helpful to me.

rreddy78, 2023-01-27:
Hello @FrankZTH :

Wonder if you have got these compilation errors when building for 4.26 ?
```
/home/sxv1kor/Downloads/GitHub/UnrealEngine_4.26/Engine/Source/Runtime/Engine/Private/CubemapUnwrapUtils.cpp:298:32: error: use of undeclared identifier 'GetVertexShader'; did you mean 'FRHIVertexShader'?
    SetShaderValue(RHICmdList, GetVertexShader(), Transform, TransformValue);
                               ^
/home/sxv1kor/Downloads/GitHub/UnrealEngine_4.26/Engine/Source/Runtime/RHI/Public/RHIResources.h:490:7: note: 'FRHIVertexShader' declared here
class FRHIVertexShader : public FRHIGraphicsShader
      ^
/home/sxv1kor/Downloads/GitHub/UnrealEngine_4.26/Engine/Source/Runtime/Engine/Private/CubemapUnwrapUtils.cpp:313:37: error: use of undeclared identifier 'GetPixelShader'; did you mean 'FRHIPixelShader'?
    SetTextureParameter(RHICmdList, GetPixelShader(),CubeTexture,CubeTextureSampler,Texture);
                                    ^
/home/sxv1kor/Downloads/GitHub/UnrealEngine_4.26/Engine/Source/Runtime/RHI/Public/RHIResources.h:508:7: note: 'FRHIPixelShader' declared here
class FRHIPixelShader : public FRHIGraphicsShader
      ^
/home/sxv1kor/Downloads/GitHub/UnrealEngine_4.26/Engine/Source/Runtime/Engine/Private/CubemapUnwrapUtils.cpp:316:32: error: use of undeclared identifier 'GetPixelShader'; did you mean 'FRHIPixelShader'?
    SetShaderValue(RHICmdList, GetPixelShader(), PackedProperties0, PackedProperties0Value);
                               ^
/home/sxv1kor/Downloads/GitHub/UnrealEngine_4.26/Engine/Source/Runtime/RHI/Public/RHIResources.h:508:7: note: 'FRHIPixelShader' declared here
class FRHIPixelShader : public FRHIGraphicsShader
      ^
/home/sxv1kor/Downloads/GitHub/UnrealEngine_4.26/Engine/Source/Runtime/Engine/Private/CubemapUnwrapUtils.cpp:317:32: error: use of undeclared identifier 'GetPixelShader'; did you mean 'FRHIPixelShader'?
    SetShaderValue(RHICmdList, GetPixelShader(), ColorWeights, ColorWeightsValue);
                               ^
/home/sxv1kor/Downloads/GitHub/UnrealEngine_4.26/Engine/Source/Runtime/RHI/Public/RHIResources.h:508:7: note: 'FRHIPixelShader' declared here
class FRHIPixelShader : public FRHIGraphicsShader
      ^
/home/sxv1kor/Downloads/GitHub/UnrealEngine_4.26/Engine/Source/Runtime/Engine/Private/CubemapUnwrapUtils.cpp:318:32: error: use of undeclared identifier 'GetPixelShader'; did you mean 'FRHIPixelShader'?
    SetShaderValue(RHICmdList, GetPixelShader(), Gamma, GammaValue);
                               ^
/home/sxv1kor/Downloads/GitHub/UnrealEngine_4.26/Engine/Source/Runtime/RHI/Public/RHIResources.h:508:7: note: 'FRHIPixelShader' declared here
class FRHIPixelShader : public FRHIGraphicsShader
      ^
/home/sxv1kor/Downloads/GitHub/UnrealEngine_4.26/Engine/Source/Runtime/Engine/Private/CubemapUnwrapUtils.cpp:320:32: error: use of undeclared identifier 'GetPixelShader'; did you mean 'FRHIPixelShader'?
    SetShaderValue(RHICmdList, GetPixelShader(), DistortionCoeffs, DistortionCoeffsValue);
                               ^
/home/sxv1kor/Downloads/GitHub/UnrealEngine_4.26/Engine/Source/Runtime/RHI/Public/RHIResources.h:508:7: note: 'FRHIPixelShader' declared here
class FRHIPixelShader : public FRHIGraphicsShader
      ^
/home/sxv1kor/Downloads/GitHub/UnrealEngine_4.26/Engine/Source/Runtime/Engine/Private/CubemapUnwrapUtils.cpp:321:32: error: use of undeclared identifier 'GetPixelShader'; did you mean 'FRHIPixelShader'?
    SetShaderValue(RHICmdList, GetPixelShader(), CameraMatrix, CameraMatrixValue);
                               ^
/home/sxv1kor/Downloads/GitHub/UnrealEngine_4.26/Engine/Source/Runtime/RHI/Public/RHIResources.h:508:7: note: 'FRHIPixelShader' declared here
class FRHIPixelShader : public FRHIGraphicsShader
      ^
/home/sxv1kor/Downloads/GitHub/UnrealEngine_4.26/Engine/Source/Runtime/Engine/Private/CubemapUnwrapUtils.cpp:324:32: error: use of undeclared identifier 'GetPixelShader'; did you mean 'FRHIPixelShader'?
    SetShaderValue(RHICmdList, GetPixelShader(), ImageSize, ImageSizeF);
                               ^
/home/sxv1kor/Downloads/GitHub/UnrealEngine_4.26/Engine/Source/Runtime/RHI/Public/RHIResources.h:508:7: note: 'FRHIPixelShader' declared here
class FRHIPixelShader : public FRHIGraphicsShader
      ^
/home/sxv1kor/Downloads/GitHub/UnrealEngine_4.26/Engine/Source/Runtime/Engine/Private/CubemapUnwrapUtils.cpp:325:32: error: use of undeclared identifier 'GetPixelShader'; did you mean 'FRHIPixelShader'?
    SetShaderValue(RHICmdList, GetPixelShader(), MaxAngle, MaxAngleValue);
                               ^
/home/sxv1kor/Downloads/GitHub/UnrealEngine_4.26/Engine/Source/Runtime/RHI/Public/RHIResources.h:508:7: note: 'FRHIPixelShader' declared here
class FRHIPixelShader : public FRHIGraphicsShader
      ^
/home/sxv1kor/Downloads/GitHub/UnrealEngine_4.26/Engine/Source/Runtime/Engine/Private/CubemapUnwrapUtils.cpp:349:80: error: indirection requires pointer operand ('TShaderMapRef<FCubemapTexturePropertiesVSFisheye>' invalid)
    GraphicsPSOInit.BoundShaderState.VertexShaderRHI = GETSAFERHISHADER_VERTEX(*VertexShader);
                                                                               ^~~~~~~~~~~~~
/home/sxv1kor/Downloads/GitHub/UnrealEngine_4.26/Engine/Source/Runtime/Engine/Private/CubemapUnwrapUtils.cpp:350:78: error: indirection requires pointer operand ('TShaderMapRef<FCubemapTexturePropertiesPSFisheye<true> >' invalid)
[... cut ...]
                                                                             ^~~~~~~~~~~~
/home/sxv1kor/Downloads/GitHub/UnrealEngine_4.26/Engine/Source/Runtime/Engine/Private/CubemapUnwrapUtils.cpp:336:9: note: in instantiation of function template specialization 'FMipLevelBatchedElementParametersFisheye::BindShaders<FCubemapTexturePropertiesPSFisheye<false> >' requested here
        BindShaders<FCubemapTexturePropertiesPSFisheye<false> >(RHICmdList, GraphicsPSOInit, InFeatureLevel, InTransform, InGamma, ColorWeights, Texture);
        ^
```

dongkeyan, 2023-02-15:
> Hello @FrankZTH :
> 
> Wonder if you have got these compilation errors when building for 4.26 ?
> 
> ```
> /home/sxv1kor/Downloads/GitHub/UnrealEngine_4.26/Engine/Source/Runtime/Engine/Private/CubemapUnwrapUtils.cpp:298:32: error: use of undeclared identifier 'GetVertexShader'; did you mean 'FRHIVertexShader'?
>     SetShaderValue(RHICmdList, GetVertexShader(), Transform, TransformValue);
>                                ^
> /home/sxv1kor/Downloads/GitHub/UnrealEngine_4.26/Engine/Source/Runtime/RHI/Public/RHIResources.h:490:7: note: 'FRHIVertexShader' declared here
> class FRHIVertexShader : public FRHIGraphicsShader
>       ^
> /home/sxv1kor/Downloads/GitHub/UnrealEngine_4.26/Engine/Source/Runtime/Engine/Private/CubemapUnwrapUtils.cpp:313:37: error: use of undeclared identifier 'GetPixelShader'; did you mean 'FRHIPixelShader'?
>     SetTextureParameter(RHICmdList, GetPixelShader(),CubeTexture,CubeTextureSampler,Texture);
>                                     ^
> ...
> /home/sxv1kor/Downloads/GitHub/UnrealEngine_4.26/Engine/Source/Runtime/Engine/Private/CubemapUnwrapUtils.cpp:349:80: error: indirection requires pointer operand ('TShaderMapRef<FCubemapTexturePropertiesVSFisheye>' invalid)
>     GraphicsPSOInit.BoundShaderState.VertexShaderRHI = GETSAFERHISHADER_VERTEX(*VertexShader);
>                                                                                ^~~~~~~~~~~~~
> /home/sxv1kor/Downloads/GitHub/UnrealEngine_4.26/Engine/Source/Runtime/Engine/Private/CubemapUnwrapUtils.cpp:350:78: error: indirection requires pointer operand ('TShaderMapRef<FCubemapTexturePropertiesPSFisheye<true> >' invalid)
>     GraphicsPSOInit.BoundShaderState.PixelShaderRHI = GETSAFERHISHADER_PIXEL(*PixelShader);
>                                                                              ^~~~~~~~~~~~
> /home/sxv1kor/Downloads/GitHub/UnrealEngine_4.26/Engine/Source/Runtime/Engine/Private/CubemapUnwrapUtils.cpp:332:9: note: in instantiation of function template specialization 'FMipLevelBatchedElementParametersFisheye::BindShaders<FCubemapTexturePropertiesPSFisheye<true> >' requested here
>         BindShaders<FCubemapTexturePropertiesPSFisheye<true> >(RHICmdList, GraphicsPSOInit, InFeatureLevel, InTransform, InGamma, ColorWeights, Texture);
>         ^
> /home/sxv1kor/Downloads/GitHub/UnrealEngine_4.26/Engine/Source/Runtime/Engine/Private/CubemapUnwrapUtils.cpp:350:78: error: indirection requires pointer operand ('TShaderMapRef<FCubemapTexturePropertiesPSFisheye<false> >' invalid)
>     GraphicsPSOInit.BoundShaderState.PixelShaderRHI = GETSAFERHISHADER_PIXEL(*PixelShader);
>                                                                              ^~~~~~~~~~~~
> /home/sxv1kor/Downloads/GitHub/UnrealEngine_4.26/Engine/Source/Runtime/Engine/Private/CubemapUnwrapUtils.cpp:336:9: note: in instantiation of function template specialization 'FMipLevelBatchedElementParametersFisheye::BindShaders<FCubemapTexturePropertiesPSFisheye<false> >' requested here
>         BindShaders<FCubemapTexturePropertiesPSFisheye<false> >(RHICmdList, GraphicsPSOInit, InFeatureLevel, InTransform, InGamma, ColorWeights, Texture);
>         ^
> ```

@rreddy78 
Hi，How did you solve the problem？

rreddy78, 2023-02-15:
> @rreddy78 Hi，How did you solve the problem？

Use
RHICmdList.GetBoundVertexShader() instead of GetVertexShader()
RHICmdList.GetBoundPixelShader() instead of GetPixelShader()

Instead of the given macros, use code as below.
GraphicsPSOInit.BoundShaderState.VertexShaderRHI = VertexShader.GetVertexShader();
GraphicsPSOInit.BoundShaderState.PixelShaderRHI = PixelShader.GetPixelShader();

JosenJiang, 2023-03-29:
@FrankZTH Hello, Have you solved the problem? Would you mind sharing your method？

AHAHA124, 2023-04-20:
> @FrankZTH Hello, Have you solved the problem? Would you mind sharing your method？

Hello, Have you solved the problem? 
I tried the method @rreddy78  said
I use
RHICmdList.GetBoundVertexShader() instead of GetVertexShader()
RHICmdList.GetBoundPixelShader() instead of GetPixelShader()

but i dont konw how to 'Instead of the given macros, use code as below.'
Would you mind sharing your method？

AHAHA124, 2023-04-21:
@JosenJiang 
Hello, Have you solved the problem?
I tried @rreddy78 method,

Use
RHICmdList.GetBoundVertexShader() instead of GetVertexShader()
RHICmdList.GetBoundPixelShader() instead of GetPixelShader()
Instead of the given macros, use code as below.
GraphicsPSOInit.BoundShaderState.VertexShaderRHI = VertexShader.GetVertexShader();
GraphicsPSOInit.BoundShaderState.PixelShaderRHI = PixelShader.GetPixelShader();

But there is still 'error: corrupt patch at line 99'.
Would you mind sharing your method？

Hassan-BABAOUSMAIL, 2023-04-25:
> @JosenJiang Hello, Have you solved the problem? I tried @rreddy78 method,
>
> Use RHICmdList.GetBoundVertexShader() instead of GetVertexShader() RHICmdList.GetBoundPixelShader() instead of [...]

You can apply it manually. You can check this link: https://stackoverflow.com/questions/987372/what-is-the-format-of-a-patch-file

yunhao666, 2023-07-19:
> We would have to comment out two instances of the member function: virtual bool Serialize(FArchive& Ar) override in the file Engine/Source/Runtime/Engine/Public/CubemapUnwrapUtils.h and it should compile for 4.26.
> 
> Though the patch file seems corrupted and I could not apply it automatically.

How to solve this error？

little-black-sjtu, 2023-07-22:
Hello, sorry to bother you!
I installed UE 4.26 on my Windows 11.
Then, I would like to add annaornatskaya's fisheye PR for my research purpose.
Therefore, I modified the  `CubemapUnwrapUtils.cpp` and `CubemapUnwrapUtils.h` exactly as you mentioned, then I rebuilt the UE through UE4.sln.
Everything looked fine, but after I clicked the `UE4Editor.exe` in the folder 'Engine/Binaries/Win64' to launch the engine, it showed 应用程序无法正常启动.
In order to know the reason, I deleted the extra code and rebuilt it again, everything worked fine. But I really need the fisheye sensor for my research, how can I solve it?
Thank you very much!

tngh5004, 2023-08-13:
> 

> 

Hi @rreddy78,
Your answer is very helpful. However, I am not sure how to solve this.
"We would have to comment out two instances of the member function:
virtual bool Serialize(FArchive& Ar) override in the file Engine/Source/Runtime/Engine/Public/CubemapUnwrapUtils.h and it should compile for 4.26.

Though the patch file seems corrupted and I could not apply it automatically."
and this one too.
"Do you know what this build error that I am getting could be caused due to ?

Assertion failed: AlignedOffset == TypeLayout.Size [File:/home/sxv1kor/Downloads/GitHub/UnrealEngine_4.26/Engine/Source/Runtime/Core/Private/Serialization/MemoryImage.cpp] [Line: 224]
  [FCubemapTexturePropertiesVSFisheye] Calculated Size: 280, Real Size: 288
Took 0.942226s to run UE4Editor, ExitCode=133
EIDT:

Might be due to the lack of LAYOUT_FIELD macros:
https://forums.unrealengine.com/t/assertion-failed-alignedoffset-typelayout-size/243865"

i just change the CubemapUnwapUtils.h code
private:
	FShaderParameter Transform;
};
>>>>>>>>>>>>>>>>
private:
	LAYOUT_FIELD(FShaderParameter, Transform);
};

but it's not working...

EbyGunner, 2024-02-16:
/data/UnrealEngine_4.26/Engine/Source/Runtime/Renderer/Private/SceneCaptureRendering.cpp:941:6: error: no matching function for call to 'UpdateSceneCaptureContent_RenderThread'
                                        UpdateSceneCaptureContent_RenderThread(RHICmdList, SceneRenderer, TextureRenderTarget, TextureRenderTarget, EventName, FResolveParams(FResolveRect(), TargetFace), false, FGenerateMipsParams());
                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/data/UnrealEngine_4.26/Engine/Source/Runtime/Renderer/Private/SceneCaptureRendering.cpp:379:13: note: candidate function not viable: requires 10 arguments, but 8 were provided
static void UpdateSceneCaptureContent_RenderThread(

How to solve this error?

Tefeink, 2025-04-01:
> /data/UnrealEngine_4.26/Engine/Source/Runtime/Renderer/Private/SceneCaptureRendering.cpp:941:6: error: no matching function for call to 'UpdateSceneCaptureContent_RenderThread'/data/UnrealEngine_4.26/Engine/Source/Runtime/Renderer/Private/SceneCaptureRendering.cpp:941:6: 错误：没有匹配的函数来调用‘UpdateSceneCaptureContent_RenderThread’ UpdateSceneCaptureContent_RenderThread(RHICmdList, SceneRenderer, TextureRenderTarget, TextureRenderTarget, EventName, FResolveParams(FResolveRect(), TargetFace), false, FGenerateMipsParams());UpdateSceneCaptureContent_RenderThread(RHICmdList、SceneRenderer、TextureRenderTarget、TextureRenderTarget、EventName、FResolveParams(FResolveRect()、TargetFace)、false、FGenerateMipsParams()); ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ /data/UnrealEngine_4.26/Engine/Source/Runtime/Renderer/Private/SceneCaptureRendering.cpp:379:13: note: candidate function not viable: requires 10 arguments, but 8 were provided/data/UnrealEngine_4.26/Engine/Source/Runtime/Renderer/Private/SceneCaptureRendering.cpp:379:13: 注意：候选函数不可行：需要 10 个参数，但提供了 8 个 static void UpdateSceneCaptureContent_RenderThread(静态 void UpdateSceneCaptureContent_RenderThread（ ^ 1 error generated.  生成 1 个错误。 make[1]: *** [Makefile:1080: CarlaUE4Editor] Error 6make[1]: *** [Makefile:1080: CarlaUE4Editor] 错误 6 make[1]: Leaving directory '/home/sim_user/carla/carla/Unreal/CarlaUE4'make[1]: 离开目录 '/home/sim_user/carla/carla/Unreal/CarlaUE4' make: *** [Util/BuildTools/Linux.mk:7: launch] Error 2make：*** [Util/BuildTools/Linux.mk:7: launch] 错误 2
> 
> How to solve this error?如何解决这个错误？

UnrealEngine_4.26 don't need to change file [SceneCaptureRendering.cpp], keep the source file.
change other 3 file and make launch.