carla-simulator/carla #5681: How do I import newly added sensors into a Carla Package

hashim19, 2022-08-19:
Hey Guys, 
I have added a new lidar sensor in carla. This sensor is working in the source carla i.e. when I do make launch and run the `python3 manual_control.py` (I can see its data in rviz). Then, I went on to package this sensor using the command `make package`, which creates a new directory Dist and a packaged carla inside it. My Dist folder looks like this, 
[image]

I went inside CARLA_Shipping_e0e00321-dirty/LinuxNoEditor/ directrory, and in one terminal I ran the carla server using `./CarlaUE4.sh` and in anoher terminal I ran a python client and spawned my new lidar in it. But I am getting the segmentation fault error. Look at video below,
[sensor testing in packaged carla](https://drive.google.com/file/d/1shcb1EmQ1_r5HtA1oV4OUSSt4baZCnsg/view?usp=sharing)

I thought there maybe an issue with my new lidar sensor implementation (though it is working with source build carla). I went on to check the safe distance sensor as described here [create new sensor tutorial](https://carla.readthedocs.io/en/0.9.13/tuto_D_create_sensor/). I packaged it using `make package` and ran the server and the python client but I am getting the same segmentation fault error. 

Can anyone help me with this? 

**Note:** 
Before this I have also added a new vehicle and its working perfectly fine in the packaged version of Carla. so, I am not sure whats going wrong here. 

CARLA version: 0.9.13
Platform/OS: Ubuntu 18
Problem you have experienced: new sensor added not working in a packaged carla
What you expected to happen: the newly added sensor should be working in the packaged carla
Steps to reproduce: Do `make package` in source carla and try to test the newly added sensor in the packaged carla.
Other information (documentation you consulted, workarounds you tried):

ccqil, 2022-08-27:
[image]
You are using the official python package, but your new sensor API doesn't exist in it. So you will get a segmentation error when you try to use it.
You should use the egg or whl file build by yourself, not the official package.

Blyron, 2026-11-12:
This is about packaging on Unreal Engine 4. In CARLA 1.0 (Unreal Engine 5.8, branch `ue58-dev`) new sensors are added as in [Create a sensor](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/dev_create_sensor.md) and the package is built with the CMake `package` target. Closing as obsolete. If it still happens there, please open a new issue with the steps to reproduce and the logs.

closed 2026-11-12 by Blyron