carla-simulator/carla #5683: Deploying Carla Docker on a Windows machine

prasilva98, 2022-08-22:
Hi there, 
Is it possible to run Carla from Docker in a windows machine? Currently when I follow the installation guide it seems to only reference running it from a Linux machine, and, when I try to run the docker container after pulling it I get the following error, which seems to be linux related:
```
 docker run --privileged --net=host -e DISPLAY=$DISPLAY carlasim/carla:latest /bin/bash ./CarlaUE4.sh
sh: 1: xdg-user-dir: not found
error: XDG_RUNTIME_DIR not set in the environment.
error: XDG_RUNTIME_DIR not set in the environment.
error: XDG_RUNTIME_DIR not set in the environment.

Thank you for your time!

EunhoCho, 2022-09-28:
Same problem on my machine..

JasonBourne1998, 2022-11-17:
I get the same problem. @EunhoCho @prasilva98 do you solve this problem?

koking0, 2023-01-04:
me to, but OS is Ubuntu 20.04.

crazyqipython, 2023-02-07:
Any progress on this issue? Thanks.

koking0, 2023-02-07:
这是来自QQ邮箱的假期自动回复邮件。你好，我最近正在休假中，无法亲自回复你的邮件。我将在看见后，尽快给你回复。

aronfeher, 2023-03-11:
I've got this issue too. Any news?

merryHunter, 2024-09-01:
anything was ever discovered related to this issue?

Blyron, 2026-11-12:
The CARLA Docker images are for Linux hosts (see [Running CARLA in Docker](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/build_docker.md)). On Windows use the native package. Closing.

closed 2026-11-12 by Blyron