carla-simulator/carla #5686: Trying a Windows 10 Build of Carla (0.9.13) - Getting Fatal error when trying to make PythonAPI

mspring6, 2022-08-22:
Output from console: 

client::World::GetWeather(void)const " (?GetWeather@World@client@carla@@QBE?AVWeatherParameters@rpc@3@XZ)
libcarla.obj : error LNK2001: unresolved external symbol "public: unsigned __int64 __thiscall carla::client::World::ApplySettings(class carla::rpc::EpisodeSettings const &,class carla::time_duration)" (?ApplySettings@World@client@carla@@QAE_KABVEpisodeSettings@rpc@3@Vtime_duration@3@@Z)
libcarla.obj : error LNK2001: unresolved external symbol "public: class carla::rpc::EpisodeSettings __thiscall carla::client::World::GetSettings(void)const " (?GetSettings@World@client@carla@@QBE?AVEpisodeSettings@rpc@3@XZ)
libcarla.obj : error LNK2001: unresolved external symbol "public: class boost::shared_ptr<class carla::client::Actor> __thiscall carla::client::World::GetSpectator(void)const " (?GetSpectator@World@client@carla@@QBE?AV?$shared_ptr@VActor@client@carla@@@boost@@XZ)
libcarla.obj : error LNK2001: unresolved external symbol "public: class boost::optional<class carla::geom::Location> __thiscall carla::client::World::GetRandomLocationFromNavigation(void)const " (?GetRandomLocationFromNavigation@World@client@carla@@QBE?AV?$optional@VLocation@geom@carla@@@boost@@XZ)
libcarla.obj : error LNK2001: unresolved external symbol "public: class std::vector<struct std::pair<unsigned int,unsigned int>,class std::allocator<struct std::pair<unsigned int,unsigned int> > > __thiscall carla::client::World::GetVehiclesLightStates(void)const " (?GetVehiclesLightStates@World@client@carla@@QBE?AV?$vector@U?$pair@II@std@@V?$allocator@U?$pair@II@std@@@2@@std@@XZ)
libcarla.obj : error LNK2001: unresolved external symbol "public: class boost::shared_ptr<class carla::client::BlueprintLibrary> __thiscall carla::client::World::GetBlueprintLibrary(void)const " (?GetBlueprintLibrary@World@client@carla@@QBE?AV?$shared_ptr@VBlueprintLibrary@client@carla@@@boost@@XZ)
libcarla.obj : error LNK2001: unresolved external symbol "public: void __thiscall carla::client::World::UnloadLevelLayer(enum carla::rpc::MapLayer)const " (?UnloadLevelLayer@World@client@carla@@QBEXW4MapLayer@rpc@3@@Z)
libcarla.obj : error LNK2001: unresolved external symbol "public: void __thiscall carla::client::World::LoadLevelLayer(enum carla::rpc::MapLayer)const " (?LoadLevelLayer@World@client@carla@@QBEXW4MapLayer@rpc@3@@Z)
libcarla.obj : error LNK2001: unresolved external symbol "public: class boost::shared_ptr<class carla::client::Map> __thiscall carla::client::World::GetMap(void)const " (?GetMap@World@client@carla@@QBE?AV?$shared_ptr@VMap@client@carla@@@boost@@XZ)
libcarla.obj : error LNK2001: unresolved external symbol "public: static class carla::rpc::WeatherParameters carla::rpc::WeatherParameters::SoftRainSunset" (?SoftRainSunset@WeatherParameters@rpc@carla@@2V123@A)
libcarla.obj : error LNK2001: unresolved external symbol "public: static class carla::rpc::WeatherParameters carla::rpc::WeatherParameters::Default" (?Default@WeatherParameters@rpc@carla@@2V123@A)
libcarla.obj : error LNK2001: unresolved external symbol "public: static class carla::rpc::WeatherParameters carla::rpc::WeatherParameters::MidRainyNoon" (?MidRainyNoon@WeatherParameters@rpc@carla@@2V123@A)
libcarla.obj : error LNK2001: unresolved external symbol "public: static class carla::rpc::WeatherParameters carla::rpc::WeatherParameters::SoftRainNoon" (?SoftRainNoon@WeatherParameters@rpc@carla@@2V123@A)
libcarla.obj : error LNK2001: unresolved external symbol "public: static class carla::rpc::WeatherParameters carla::rpc::WeatherParameters::WetCloudySunset" (?WetCloudySunset@WeatherParameters@rpc@carla@@2V123@A)
libcarla.obj : error LNK2001: unresolved external symbol "public: static class carla::rpc::WeatherParameters carla::rpc::WeatherParameters::WetCloudyNight" (?WetCloudyNight@WeatherParameters@rpc@carla@@2V123@A)
libcarla.obj : error LNK2001: unresolved external symbol "public: static class carla::rpc::WeatherParameters carla::rpc::WeatherParameters::MidRainyNight" (?MidRainyNight@WeatherParameters@rpc@carla@@2V123@A)
libcarla.obj : error LNK2001: unresolved external symbol "public: static class carla::rpc::WeatherParameters carla::rpc::WeatherParameters::HardRainNoon" (?HardRainNoon@WeatherParameters@rpc@carla@@2V123@A)
libcarla.obj : error LNK2001: unresolved external symbol "public: static class carla::rpc::WeatherParameters carla::rpc::WeatherParameters::CloudySunset" (?CloudySunset@WeatherParameters@rpc@carla@@2V123@A)
libcarla.obj : error LNK2001: unresolved external symbol "public: static class carla::rpc::WeatherParameters carla::rpc::WeatherParameters::ClearNight" (?ClearNight@WeatherParameters@rpc@carla@@2V123@A)
libcarla.obj : error LNK2001: unresolved external symbol "public: static class carla::rpc::WeatherParameters carla::rpc::WeatherParameters::HardRainSunset" (?HardRainSunset@WeatherParameters@rpc@carla@@2V123@A)
libcarla.obj : error LNK2001: unresolved external symbol "public: static class carla::rpc::WeatherParameters carla::rpc::WeatherParameters::MidRainSunset" (?MidRainSunset@WeatherParameters@rpc@carla@@2V123@A)
libcarla.obj : error LNK2001: unresolved external symbol "public: static class carla::rpc::WeatherParameters carla::rpc::WeatherParameters::WetNoon" (?WetNoon@WeatherParameters@rpc@carla@@2V123@A)
libcarla.obj : error LNK2001: unresolved external symbol "public: static class carla::rpc::WeatherParameters carla::rpc::WeatherParameters::ClearNoon" (?ClearNoon@WeatherParameters@rpc@carla@@2V123@A)
libcarla.obj : error LNK2001: unresolved external symbol "public: static class carla::rpc::WeatherParameters carla::rpc::WeatherParameters::ClearSunset" (?ClearSunset@WeatherParameters@rpc@carla@@2V123@A)
libcarla.obj : error LNK2001: unresolved external symbol "public: static class carla::rpc::WeatherParameters carla::rpc::WeatherParameters::SoftRainNight" (?SoftRainNight@WeatherParameters@rpc@carla@@2V123@A)
libcarla.obj : error LNK2001: unresolved external symbol "public: static class carla::rpc::WeatherParameters carla::rpc::WeatherParameters::CloudyNight" (?CloudyNight@WeatherParameters@rpc@carla@@2V123@A)
libcarla.obj : error LNK2001: unresolved external symbol "public: static class carla::rpc::WeatherParameters carla::rpc::WeatherParameters::WetSunset" (?WetSunset@WeatherParameters@rpc@carla@@2V123@A)
libcarla.obj : error LNK2001: unresolved external symbol "public: static class carla::rpc::WeatherParameters carla::rpc::WeatherParameters::WetNight" (?WetNight@WeatherParameters@rpc@carla@@2V123@A)
libcarla.obj : error LNK2001: unresolved external symbol "public: static class carla::rpc::WeatherParameters carla::rpc::WeatherParameters::WetCloudyNoon" (?WetCloudyNoon@WeatherParameters@rpc@carla@@2V123@A)
libcarla.obj : error LNK2001: unresolved external symbol "public: static class carla::rpc::WeatherParameters carla::rpc::WeatherParameters::HardRainNight" (?HardRainNight@WeatherParameters@rpc@carla@@2V123@A)
libcarla.obj : error LNK2001: unresolved external symbol "public: static class carla::rpc::WeatherParameters carla::rpc::WeatherParameters::CloudyNoon" (?CloudyNoon@WeatherParameters@rpc@carla@@2V123@A)
libcarla.obj : error LNK2001: unresolved external symbol "public: static class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const __cdecl carla::road::SignalType::RecomendedSpeedEnd(void)" (?RecomendedSpeedEnd@SignalType@road@carla@@SA?BV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@XZ)
libcarla.obj : error LNK2001: unresolved external symbol "public: static class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const __cdecl carla::road::SignalType::RecomendedSpeed(void)" (?RecomendedSpeed@SignalType@road@carla@@SA?BV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@XZ)
libcarla.obj : error LNK2001: unresolved external symbol "public: static class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const __cdecl carla::road::SignalType::DeadEnd(void)" (?DeadEnd@SignalType@road@carla@@SA?BV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@XZ)
libcarla.obj : error LNK2001: unresolved external symbol "public: static class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const __cdecl carla::road::SignalType::Highway(void)" (?Highway@SignalType@road@carla@@SA?BV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@XZ)
[... cut ...]
libcarla.obj : error LNK2001: unresolved external symbol _png_write_end
libcarla.obj : error LNK2001: unresolved external symbol _png_set_compression_window_bits
build\lib.win32-3.7\carla\libcarla.cp37-win32.pyd : fatal error LNK1120: 402 unresolved externals
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x86\\link.exe' failed with exit status 1120

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "C:\carla\PythonAPI\carla\dist"!

qbe, 2022-08-22:
please _always_ post code or error logs inside code blocks

`like this`

```
or like this
multiline code block
```

to avoid spamming users with short usernames

brscholz, 2022-09-02:
Hi @mspring6,

this is a very common error that occurs when you have more than one Python version on your machine. Make sure that the commands _py -3_, _py_ and _python_ in your command line all call the same Python version. If this is not the case you may want to replace all occurences of _py_ in the Carla build scripts with _python_.

Snju1212, 2022-09-08:
Hva you solved that because I am getting another error while executing make PythonAPI.

-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --all
'where' is not recognized as an internal or external command,
operable program or batch file.

-[Setup]: [ERROR] Can't find Visual Studio compiler (cl.exe).
          [ERROR] Possible causes:
          [ERROR]  - Make sure you use x64 (not x64_x86!)
          [ERROR]  - You are not using "Visual Studio x64 Native Tools Command Prompt".

I am using X64 Native Tools Command Prompt for VS 2019 for carla 0.9.13.

kmsgm, 2024-03-06:
> Hi @mspring6,
>
> this is a very common error that occurs when you have more than one Python version on your machine. Make sure that the commands _py -3_, _py_ and _python_ in your command line all [...]

I have only one python version but I got still this error

```
C:\>py --list
 -V:3.7 *         Python 3.7 (64-bit)
```

Blyron, 2026-11-12:
This is an error from the old Make/`.bat` build of CARLA 0.9.x on Unreal Engine 4. That build system is gone: CARLA 1.0 (Unreal Engine 5.8, branch `ue58-dev`) builds with CMake, which downloads and builds the dependencies itself. See [Build on Windows](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/build_windows_ue5.md). Closing as obsolete. If the current build fails for you, please open a new issue with the full log.

closed 2026-11-12 by Blyron