carla-simulator/carla #5689: Custom map with Props not loaded properly on docker image

sakuragi97, 2022-08-23:
CARLA version: 0.9.13
Platform/OS: Ubuntu 20.04.4 LTS
Problem you have experienced:

I used to create custom maps with Roadrunner and import and run them on Unreal (on Windows on my local machine) and they work fine.
Then I needed to run them on a server and run Carla in the background without rendering, this is why I proceed with the solution of building Carla image with docker with the new custom maps (Ingest the map files into packages and then import them in carla).
So I used to go through all this process with different maps that I made and they all work fine on the server. The problem now is, I made a new map with a couple of elements in it (Trees, Buildings, vehicles...), it works on Unreal (on Windows on my local machine), and the process of exporting and ingesting and importing the map files is done without any error, however, when I try to run the simulation (manual_control script for example) on the server after ingesting, the vehicle is spawned without a map, which keeps the vehicle falling down.

Images of simulation of the map that is not working on server:
[image]

Some images of the map on Unreal:
[image]

[image]

Image of previous maps that works fine on server:
[image]
[image]

I'm not sure if the map is not working because of the number of elements in it?
(default Carla maps are usually more complex than this one)
(I think this map is not considered large since it is 360m x 390m)

sakuragi97, 2022-09-20:
Update: after reducing the number of objects on the map (reducing trees, big buildings) the map works fine on the server after ingesting. The question here is how many elements you can add to the map so the ingestion process can works fine. (maybe how many objects per square meter?)

sakuragi97, 2022-09-20:
Also, I notice one thing that there are quite differences between the graphics of the objects before and after map ingestion

Blyron, 2026-11-12:
This is about the Unreal Engine 4 Docker ingest pipeline. In CARLA 1.0 (Unreal Engine 5.8, branch `ue58-dev`) custom maps with props are shipped as content packs, see [Content packs](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/content_authoring_packs.md). Closing as obsolete.

closed 2026-11-12 by Blyron