carla-simulator/carla #5690: How to properly export Roadrunner map from windows to linux

AurelAvisto, 2022-08-24:
CARLA version: 0.9.12
Platform/OS: Windows 11 (RoadRunner) , Ubuntu 20.04 (CARLA) x86_64
Problem you have experienced:

I tried to create some maps in RoadRunner on my windows computer then following some tutorials ( https://www.mathworks.com/help/roadrunner/ug/export-to-carla.html, https://carla.readthedocs.io/en/0.9.12/tuto_M_add_map_source/, https://carla.readthedocs.io/en/latest/tuto_M_add_map_alternative) I export them into the forked version of UE4editor 4.26. In unreal I can start manual_control.py in order to test my maps. My problem is that I use a dockerized version of Carla that run on another linux computer and I didn't succeed in exporting/packaging my maps from windows to Linux.

Using a docker image of unreal engine and the script docker_tools.py on my linux machine, I can ingest a small map into Carla but most of the textures are missing, I can't define pedestrian navigation and if I add a building into my map, the script produce an empty map.

I also tried cross-platform packaging on my windows computer targeting linux but It always fails because some static lib are missing (librcp.a , libcarla_server.a) even if I rebuild the UE4editor entirely. Maybe I missed something, the documentation for the cross-platform packaging is confusing and does not seems to be up-to-date.

On my linux, I can't run the GUI of the UnrealEngine Editor because of some problems with Vulkan, that's why I'm using unreal on windows. At this point I don't know what to try next.

Did anyone have a workflow for packaging RoadRunner map and import them into a dockerized version of Carla that run on a Linux host?

Thank you.

StephanieMussi, 2022-09-09:
Export the map in RoadRunner with "File --> Export --> Carla" option, then follow the [documentation](https://carla.readthedocs.io/en/0.9.13/tuto_M_add_map_source/) on how to import the map

AurelAvisto, 2022-09-12:
@StephanieMussi thanks, but the point is that I already tried that and in my case it did not work.

Dragonking3x, 2022-10-14:
I may can help with the empty map one.

I made a Carla package myself, and the map was empty but had ground and roads.
In my case the layer was not loaded.
I could activate it with the Python script
carla.Client("127.0.0.1", 2000).get_world().load_map_layer(carla.MapLayer.All)

Maybe that is the case for you, too

closed 2025-05-08 by Blyron