carla-simulator/carla #5691: Only the cars that are in the front view of the picture

AmirMz1, 2022-08-24:
Hi, I'm recording data, how to read vehicle are front view (not back), 
I have other vehicle location and my car location...

TR-Monash, 2022-08-25:
You can use the semantic segmentation sensor coupled with[ this pr](https://github.com/carla-simulator/carla/pull/5452) to do this bys setting 
```
self.world.set_instance_tagging_style("actor_id")
```
An example code would be: 
```
    def get_actor_ids_in_semantic_image(self):
        img = np.array(self.seg_image.raw_data)
        img = img.reshape((self.view_height, self.view_width, 4))
        actor_pixel_indices = np.unravel_index(np.in1d(img[:, :, 2], [4, 10]).nonzero()[0], img[:, :, 2].shape)
        actor_ids = img[:, :, 1][actor_pixel_indices] + (256 * img[:, :, 0][actor_pixel_indices])
        unique_actor_ids= np.unique(actor_ids)
```

closed 2022-08-26 by AmirMz1

Basavaraj-PN, 2023-04-29:
@AmirMz1  
Try this repo : [actor sensor](https://github.com/Basavaraj-PN/actor-sensor-for-carla-simulator/blob/main/README.md)