carla-simulator/carla #5693: Friction Triggers Appear to Alter The Tire Friction of *All* cars on the map

craigiedon, 2022-08-25:
**CARLA version:** 0.9.13
**Platform/OS:** Ubuntu 21.0

**Problem**: After creating a friction trigger with 0.0 friction (according to the [code in this part of the docs](https://carla.readthedocs.io/en/latest/tuto_G_add_friction_triggers/)), I spawned two vehicles with `BasicAgent` and set them to drive to next way point. When the *first* car enters the trigger zone, *both* cars spin out of control. I expected that only the car entering the trigger zone would have its wheel friction changed, but it appears to change the friction for _every_ vehicle, **even those currently outside of the trigger zone**.

I confirmed this was what was happening by running:

```
for actor in actor_list:
  wheel_phys = actor.get_physics_control().wheels[0]
  print(f"Tire Friction: {wheel_phys.tire_friction}")
```

At every tick. Which produces:

```
Time Step 35
Tire Friction: 3.5
Tire Friction: 3.5

Time Step 36
Tire Friction: 0.0
Tire Friction: 0.0
```

(The first car enters the friction trigger after time-step 35, at which point the second car is far away from the friction trigger)

Youngzipper, 2024-04-16:
> **CARLA version:** 0.9.13 **Platform/OS:** Ubuntu 21.0
>
> **Problem**: After creating a friction trigger with 0.0 friction (according to the [code in this part of the [...]
i am using friction trigger to change road friction in CARLA too. But when my ego vehicle drives into friction trigger area, the print result of tire friction is still 3.5. So I want to ask your opinion
and my 'print' code is:
        while True:
            physics = ego_vehicle.get_physics_control()
            print(f"tire friction = ", physics.wheels[0].tire_friction)
            time.sleep(0.5)

whatgit, 2024-07-11:
I have faced the same issue in CARLA version 0.9.14. It seems that once a vehicle enters the friction zone, the tire friction is changed for **all vehicles using the same blueprint** , instead of only that vehicle. It seems that it does not affect other vehicles using a different blueprint. Is this an expected behaviour for friction triggers?

Blyron, 2026-11-12:
Fixed in #10140: the friction trigger now saves and restores the friction of each vehicle that enters it, so other vehicles are no longer affected. Closing.

closed 2026-11-12 by Blyron