carla-simulator/carla #5698: Build issue Carla 0.9.13

omitheaussie, 2022-08-27:
**CARLA version:** 0.9.13
**Platform/OS:** Windows 11, I have all the  dependencies installed: make (3.81), cmake(3.24.1), python (3.8.10), Visual Studio 2019 professional, Windows 8.1 SDK, x64 VC++ Toolset, .NET framework 4.6.2
**Problem you have experienced**: make PythonAPI fails. I was already able to build Unreal Engine using Visual studio.
**Note**: I had to change this line in the install_zlib.bat 
set ZLIB_VERSION=1.2.11 
to 
set ZLIB_VERSION=1.2.12

**Steps to reproduce**: Follow the steps as per the link here: https://carla.readthedocs.io/en/latest/build_windows/
Other information (documentation you consulted, workarounds you tried): https://github.com/carla-simulator/carla/issues/4525

**Here is the output I see when make PythonAPI fails:**

osm2odr.lib(GeoConvHelper.obj) : error LNK2001: unresolved external symbol proj_create
osm2odr.lib(GeoConvHelper.obj) : error LNK2001: unresolved external symbol proj_destroy
osm2odr.lib(GeoConvHelper.obj) : error LNK2001: unresolved external symbol proj_trans
osm2odr.lib(GeoConvHelper.obj) : error LNK2001: unresolved external symbol proj_torad
osm2odr.lib(GeoConvHelper.obj) : error LNK2001: unresolved external symbol proj_todeg
osm2odr.lib(OptionsIO.obj) : error LNK2001: unresolved external symbol "public: static void * __cdecl xercesc_3_2::XMemory::operator new(unsigned __int64)" (??2XMemory@xercesc_3_2@@SAPEAX_K@Z)
osm2odr.lib(XMLSubSys.obj) : error LNK2001: unresolved external symbol "public: static void * __cdecl xercesc_3_2::XMemory::operator new(unsigned __int64)" (??2XMemory@xercesc_3_2@@SAPEAX_K@Z)
osm2odr.lib(SUMOSAXReader.obj) : error LNK2001: unresolved external symbol "public: static void * __cdecl xercesc_3_2::XMemory::operator new(unsigned __int64)" (??2XMemory@xercesc_3_2@@SAPEAX_K@Z)
osm2odr.lib(SUMOSAXReader.obj) : error LNK2001: unresolved external symbol "public: static void __cdecl xercesc_3_2::XMemory::operator delete(void *)" (??3XMemory@xercesc_3_2@@SAXPEAX@Z)
osm2odr.lib(OptionsIO.obj) : error LNK2001: unresolved external symbol "public: static void __cdecl xercesc_3_2::XMemory::operator delete(void *)" (??3XMemory@xercesc_3_2@@SAXPEAX@Z)
osm2odr.lib(XMLSubSys.obj) : error LNK2001: unresolved external symbol "public: static void __cdecl xercesc_3_2::XMemory::operator delete(void *)" (??3XMemory@xercesc_3_2@@SAXPEAX@Z)
osm2odr.lib(OptionsLoader.obj) : error LNK2001: unresolved external symbol "public: static void __cdecl xercesc_3_2::XMemory::operator delete(void *)" (??3XMemory@xercesc_3_2@@SAXPEAX@Z)
osm2odr.lib(GenericSAXHandler.obj) : error LNK2001: unresolved external symbol "public: static void __cdecl xercesc_3_2::XMemory::operator delete(void *)" (??3XMemory@xercesc_3_2@@SAXPEAX@Z)
osm2odr.lib(OptionsIO.obj) : error LNK2001: unresolved external symbol "public: __cdecl xercesc_3_2::SAXParser::SAXParser(class xercesc_3_2::XMLValidator * const,class xercesc_3_2::MemoryManager * const,class xercesc_3_2::XMLGrammarPool * const)" (??0SAXParser@xercesc_3_2@@QEAA@QEAVXMLValidator@1@QEAVMemoryManager@1@QEAVXMLGrammarPool@1@@Z)
osm2odr.lib(OptionsIO.obj) : error LNK2001: unresolved external symbol "public: virtual __cdecl xercesc_3_2::SAXParser::~SAXParser(void)" (??1SAXParser@xercesc_3_2@@UEAA@XZ)
osm2odr.lib(OptionsIO.obj) : error LNK2001: unresolved external symbol "public: void __cdecl xercesc_3_2::SAXParser::setDoNamespaces(bool)" (?setDoNamespaces@SAXParser@xercesc_3_2@@QEAAX_N@Z)
osm2odr.lib(OptionsIO.obj) : error LNK2001: unresolved external symbol "public: void __cdecl xercesc_3_2::SAXParser::setValidationScheme(enum xercesc_3_2::SAXParser::ValSchemes)" (?setValidationScheme@SAXParser@xercesc_3_2@@QEAAXW4ValSchemes@12@@Z)
osm2odr.lib(OptionsIO.obj) : error LNK2001: unresolved external symbol "public: void __cdecl xercesc_3_2::SAXParser::setDoSchema(bool)" (?setDoSchema@SAXParser@xercesc_3_2@@QEAAX_N@Z)
osm2odr.lib(OptionsIO.obj) : error LNK2001: unresolved external symbol "public: bool __cdecl xercesc_3_2::SAXParser::parseFirst(class xercesc_3_2::InputSource const &,class xercesc_3_2::XMLPScanToken &)" (?parseFirst@SAXParser@xercesc_3_2@@QEAA_NAEBVInputSource@2@AEAVXMLPScanToken@2@@Z)
osm2odr.lib(OptionsIO.obj) : error LNK2001: unresolved external symbol "public: bool __cdecl xercesc_3_2::SAXParser::parseNext(class xercesc_3_2::XMLPScanToken &)" (?parseNext@SAXParser@xercesc_3_2@@QEAA_NAEAVXMLPScanToken@2@@Z)
osm2odr.lib(OptionsIO.obj) : error LNK2001: unresolved external symbol "public: virtual void __cdecl xercesc_3_2::SAXParser::parse(char const * const)" (?parse@SAXParser@xercesc_3_2@@UEAAXQEBD@Z)
osm2odr.lib(OptionsIO.obj) : error LNK2001: unresolved external symbol "public: virtual void __cdecl xercesc_3_2::SAXParser::setDocumentHandler(class xercesc_3_2::DocumentHandler * const)" (?setDocumentHandler@SAXParser@xercesc_3_2@@UEAAXQEAVDocumentHandler@2@@Z)
osm2odr.lib(OptionsIO.obj) : error LNK2001: unresolved external symbol "public: virtual void __cdecl xercesc_3_2::SAXParser::setErrorHandler(class xercesc_3_2::ErrorHandler * const)" (?setErrorHandler@SAXParser@xercesc_3_2@@UEAAXQEAVErrorHandler@2@@Z)
osm2odr.lib(OptionsIO.obj) : error LNK2001: unresolved external symbol "public: virtual __cdecl xercesc_3_2::BinInputStream::~BinInputStream(void)" (??1BinInputStream@xercesc_3_2@@UEAA@XZ)
osm2odr.lib(SUMOSAXReader.obj) : error LNK2001: unresolved external symbol "public: virtual __cdecl xercesc_3_2::BinInputStream::~BinInputStream(void)" (??1BinInputStream@xercesc_3_2@@UEAA@XZ)
osm2odr.lib(OptionsIO.obj) : error LNK2001: unresolved external symbol "public: virtual char16_t const * __cdecl xercesc_3_2::BinInputStream::getEncoding(void)const " (?getEncoding@BinInputStream@xercesc_3_2@@UEBAPEB_SXZ)
osm2odr.lib(SUMOSAXReader.obj) : error LNK2001: unresolved external symbol "public: virtual char16_t const * __cdecl xercesc_3_2::BinInputStream::getEncoding(void)const " (?getEncoding@BinInputStream@xercesc_3_2@@UEBAPEB_SXZ)
osm2odr.lib(OptionsIO.obj) : error LNK2001: unresolved external symbol "protected: __cdecl xercesc_3_2::BinInputStream::BinInputStream(void)" (??0BinInputStream@xercesc_3_2@@IEAA@XZ)
osm2odr.lib(SUMOSAXReader.obj) : error LNK2001: unresolved external symbol "protected: __cdecl xercesc_3_2::BinInputStream::BinInputStream(void)" (??0BinInputStream@xercesc_3_2@@IEAA@XZ)
osm2odr.lib(OptionsIO.obj) : error LNK2001: unresolved external symbol "public: virtual __cdecl xercesc_3_2::InputSource::~InputSource(void)" (??1InputSource@xercesc_3_2@@UEAA@XZ)
osm2odr.lib(SUMOSAXReader.obj) : error LNK2001: unresolved external symbol "public: virtual __cdecl xercesc_3_2::InputSource::~InputSource(void)" (??1InputSource@xercesc_3_2@@UEAA@XZ)
osm2odr.lib(OptionsIO.obj) : error LNK2001: unresolved external symbol "public: virtual void __cdecl xercesc_3_2::InputSource::setEncoding(char16_t const * const)" (?setEncoding@InputSource@xercesc_3_2@@UEAAXQEB_S@Z)
osm2odr.lib(SUMOSAXReader.obj) : error LNK2001: unresolved external symbol "public: virtual void __cdecl xercesc_3_2::InputSource::setEncoding(char16_t const * const)" (?setEncoding@InputSource@xercesc_3_2@@UEAAXQEB_S@Z)
osm2odr.lib(OptionsIO.obj) : error LNK2001: unresolved external symbol "public: virtual void __cdecl xercesc_3_2::InputSource::setPublicId(char16_t const * const)" (?setPublicId@InputSource@xercesc_3_2@@UEAAXQEB_S@Z)
osm2odr.lib(SUMOSAXReader.obj) : error LNK2001: unresolved external symbol "public: virtual void __cdecl xercesc_3_2::InputSource::setPublicId(char16_t const * const)" (?setPublicId@InputSource@xercesc_3_2@@UEAAXQEB_S@Z)
osm2odr.lib(OptionsIO.obj) : error LNK2001: unresolved external symbol "public: virtual void __cdecl xercesc_3_2::InputSource::setSystemId(char16_t const * const)" (?setSystemId@InputSource@xercesc_3_2@@UEAAXQEB_S@Z)
osm2odr.lib(SUMOSAXReader.obj) : error LNK2001: unresolved external symbol "public: virtual void __cdecl xercesc_3_2::InputSource::setSystemId(char16_t const * const)" (?setSystemId@InputSource@xercesc_3_2@@UEAAXQEB_S@Z)
osm2odr.lib(OptionsIO.obj) : error LNK2001: unresolved external symbol "protected: __cdecl xercesc_3_2::InputSource::InputSource(char const * const,class xercesc_3_2::MemoryManager * const)" (??0InputSource@xercesc_3_2@@IEAA@QEBDQEAVMemoryManager@1@@Z)
osm2odr.lib(SUMOSAXReader.obj) : error LNK2001: unresolved external symbol "protected: __cdecl xercesc_3_2::InputSource::InputSource(char const * const,class xercesc_3_2::MemoryManager * const)" (??0InputSource@xercesc_3_2@@IEAA@QEBDQEAVMemoryManager@1@@Z)
[... cut ...]
osm2odr.lib(SUMOSAXAttributesImpl_Xerces.obj) : error LNK2001: unresolved external symbol "public: static void __cdecl xercesc_3_2::XMLString::release(char16_t * *,class xercesc_3_2::MemoryManager * const)" (?release@XMLString@xercesc_3_2@@SAXPEAPEA_SQEAVMemoryManager@2@@Z)
build\lib.win-amd64-3.8\carla\libcarla.cp38-win_amd64.pyd : fatal error LNK1120: 52 unresolved externals
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Professional\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\link.exe' failed with exit status 1120

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "C:\Carla\PythonAPI\carla\dist"!
-->

omitheaussie, 2022-08-27:
Aparently, during the installation of VS 2019, ".NET framework 4.6.2"  is not included. What is included is .NET framework 4.6.2 SDK and ".NET framework 4.6.2 targeting pack". This is not the same as what is required. I had to separately install this:

https://dotnet.microsoft.com/en-us/download/dotnet-framework/thank-you/net462-developer-pack-offline-installer
After installation, the problem above went away.

closed 2022-08-27 by omitheaussie

ronitsharan, 2024-05-31:
Tried this but it didn't help me 
i am still getting the same error

TroyZhangFawang, 2024-08-21:
It didn't work, how to solve this bug?