carla-simulator/carla #5699: How to get the information of actors in the field of view.

Basavaraj-PN, 2022-08-28:
Hello, is there any virtual sensor that gives actor information in certain field of view?
If it is available can we customize other senor parameter like range of the sensor and sensor position on ego.

TR-Monash, 2022-09-12:
Have a look at [this](https://github.com/carla-simulator/carla/issues/5691#issuecomment-1226833269).

glopezdiest, 2022-09-23:
Hey @Basavaraj-PN, unfortunately, CARLA doesn't have a way to do that, but the solution proposed by @TR-Monash seems to be a good one.

Closing this issue as it seems fixed, but feel free to reopen it if needed

closed 2022-09-23 by glopezdiest

Basavaraj-PN, 2023-04-29:
@glopezdiest  @TR-Monash 
hey, better solution for this issue:  [actor sensor](https://github.com/Basavaraj-PN/actor-sensor-for-carla-simulator/blob/main/README.md)