carla-simulator/carla #5707: Autopilot causing collisions

ronyshaji, 2022-08-29:
Hi

I was working with carla and tried to spawn the vehicles.
But whenever I set the vehicles in autopilot mode, it goes sideways and hit on the things.

What may be the issue ?

 world.enable_environment_objects(objects_to_toggle, True)
        
        blueprint_library = world.get_blueprint_library()
        bp = blueprint_library.filter('*vehicle*')
        # print(bp)
        
        
        spawn_points = world.get_map().get_spawn_points()
        # print(spawn_points)
        
        
        #spawn other traffic cars
        for i in range(0,50):
            autovehicle = world.try_spawn_actor(random.choice(bp),random.choice(spawn_points))
            actor_list.append(autovehicle)
            
        for vehicle in world.get_actors().filter('*vehicle*'):
            # print('created %s' % vehicle.type_id)
            vehicle.set_autopilot(True)
            
        
        #spawn an ego vehicle
        ego_vehicle = world.spawn_actor(random.choice(bp),random.choice(spawn_points))
        actor_list.append(ego_vehicle)
        ego_vehicle.set_autopilot(True)

glopezdiest, 2022-09-23:
Are you using asynchronous mode? In asynchronous mode, the time step is variable, which causes the vehicle to not correctly follow the road. 

I'd suggest switching to synchronous mode if you can

alexkutsan, 2022-10-07:
I have some kind of similar problem in synchronous mode as well, 
I want to simulate heavy traffic but, vehicles in autopilot mode behave strangely. 
Is there any limitation to the number of vehicles (I was trying 265) 
I don't see a colossal CPU load,  but I see many collisions and strange vehicles behavior. 
```
import carla
client = carla.Client('localhost', 2000)
# client.load_world('Town03')
world = client.get_world()
eagle_view = carla.Transform(carla.Location(0, 0, 320), carla.Rotation(-90, 0, 0)) 
spectator = world.get_spectator()
spectator.set_transform(eagle_view)

traffic_manager = client.get_trafficmanager()
traffic_manager.set_synchronous_mode(True)

# Set a seed so behaviour can be repeated if necessary
traffic_manager.set_random_device_seed(0)

import random 
vehicle_blueprints = world.get_blueprint_library().filter('*vehicle*')
spawn_points = world.get_map().get_spawn_points()
print(len(spawn_points))
for p in spawn_points:
    actor = world.try_spawn_actor(random.choice(vehicle_blueprints), p)
    if actor:
        traffic_manager.ignore_lights_percentage(actor, random.randint(0,50))
        actor.set_autopilot(True)
while True:
    world.tick()

```
[Screencast from 07.10.2022 19:39:24.webm](https://user-images.githubusercontent.com/860357/194617618-0cf463b4-2f83-40fe-ad69-4679724ae8b2.webm)

ronyshaji, 2022-10-28:
> I have some kind of similar problem in synchronous mode as well, I want to simulate heavy traffic but, vehicles in autopilot mode behave strangely. Is there any limitation to the number of vehicles (I was trying 265) I don't see a colossal CPU load, but I see many collisions and strange vehicles behavior.
> 
> ```python
> import carla
> client = carla.Client('localhost', 2000)
> # client.load_world('Town03')
> world = client.get_world()
> eagle_view = carla.Transform(carla.Location(0, 0, 320), carla.Rotation(-90, 0, 0)) 
> spectator = world.get_spectator()
> spectator.set_transform(eagle_view)
> 
> traffic_manager = client.get_trafficmanager()
> traffic_manager.set_synchronous_mode(True)
> 
> # Set a seed so behaviour can be repeated if necessary
> traffic_manager.set_random_device_seed(0)
> 
> import random 
> vehicle_blueprints = world.get_blueprint_library().filter('*vehicle*')
> spawn_points = world.get_map().get_spawn_points()
> print(len(spawn_points))
> for p in spawn_points:
>     actor = world.try_spawn_actor(random.choice(vehicle_blueprints), p)
>     if actor:
>         traffic_manager.ignore_lights_percentage(actor, random.randint(0,50))
>         actor.set_autopilot(True)
> while True:
>     world.tick()
> ```
> 
>  Screencast.from.07.10.2022.19.39.24.webm

Diy you able to find a work around ?

alexkutsan, 2022-10-29:
Nope :-) sad for me, but I still can't setup heavy traffic in Carla :-(

ronyshaji, 2022-10-31:
> Nope :-) sad for me, but I still can't setup heavy traffic in Carla :-(

Do you ever worked  with carla build from source ?

alexkutsan, 2022-10-31:
No, I didn't have a need for this. 
How could it help me with setting up heavy traffic?

ronyshaji, 2022-10-31:
> No, I didn't have a need for this. How could it help me with setting up heavy traffic?

No I guess the traffic is the issue of the CARLA. I still need to try heavy traffic but in my case all vehicles are going wild. Did you used the traffic manager also ?

makaveli10, 2023-05-30:
Do you have multiple clients running? In my case I had to change the traffic manager port for one of the clients so that they are not using the same port.

hexrefugee, 2023-05-30:
> 

Yes, I try 10 vehicles in tm-port 8000， and ego_vehicle with lidar in tm-port 8005. They are in different clients. But the vehicles go weird too.
I'm wondering if it's a sync issue？
The clients you made all in sync? Both client and tm?
liker that:
`
settings.synchronous_mode = True
traffic_manager.set_synchronous_mode(True)
`
Thanks for reply.

hexrefugee, 2023-05-31:
> @hexrefugee Yes for sure, everything needs to be in sync mode

Ok，I will try to in different client to set different tm-port.
Thanks for your reply!

alexkutsan, 2024-06-07:
Just curios may be someone was able to generate a heavy traffic in carla? 
My be it is easier to generate traffic with SUMO and play open scenario files in carla? 
But for this I need XODRs of carlas maps. 
Anyone know where to get it?

Blyron, 2026-11-12:
Run the world and the Traffic Manager in synchronous mode with a fixed `fixed_delta_seconds`, and use a different Traffic Manager port for each client. See [Traffic Manager](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/core_traffic_manager.md) and [Synchrony and time-step](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/core_synchrony.md). Closing. If it still happens there, please open a new issue with the steps to reproduce and the logs.

closed 2026-11-12 by Blyron