carla-simulator/carla #5708: make rebuild failing to run BuildCarlaUE4.bat

kclarke-ara, 2022-08-30:


M-Colley, 2026-10-26:
Draft PR #10023 proposes the `ue4-dev` rebuild-recipe fix: use the supported build/clean actions and preserve dependency ordering instead of forwarding an unsupported `--rebuild` flag. It includes Windows batch regressions and currently has merge conflicts. A full Unreal rebuild remains unrun.

Blyron, 2026-11-12:
This is an error from the old Make/`.bat` build of CARLA 0.9.x on Unreal Engine 4. That build system is gone: CARLA 1.0 (Unreal Engine 5.8, branch `ue58-dev`) builds with CMake, which downloads and builds the dependencies itself. See [Build on Windows](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/build_windows_ue5.md). Closing as obsolete. If the current build fails for you, please open a new issue with the full log.

closed 2026-11-12 by Blyron