carla-simulator/carla #5711: Attaching actors to each other in batch spawn

brscholz, 2022-08-31:
Hi everyone,

I'm trying to spawn actors in a batch with some actors attached to others. Using FutureActor doesn't work, probably because the SpawnActor command itself is the one creating a FutureActor?

`batch = [carla.command.SpawnActor(post_bp, trans).then(carla.command.SpawnActor(sign_bp, trans2, carla.command.FutureActor))]`

Is there another way to do it in a batch?

Best regards
Robert

qhaas, 2024-12-04:
This would be useful for spawning pedestrian actors with their controller, any progress on spawnactor chaining?

M-Colley, 2026-10-26:
Draft PR #10052 proposes the `ue58-dev` fix for this chaining pattern: a child SpawnActor with an explicit FutureActor parent is resolved to the actor just spawned. Omitted and explicit nonzero parents are preserved, including nested chains. The actual production command helper passes 7/7 native regressions on Windows and Linux, including MsgPack roundtrip and Clang sanitizers. A matching-engine hierarchy/relative-offset smoke test is included but unrun; full Unreal compilation and live attachment remain outstanding. #5814 is the same underlying root.