carla-simulator/carla #5717: Unable to edit level: Level is locked in CARLA Unreal project

snehanjalikalamkar, 2022-09-01:
I require adding some custom objects in the CARLA maps for larger towns. Whenever I drag any object into the map editor, I get an error: The requested operation could not be completed because the level is locked (see image). This does not happen with the map of Town01 and Town10. However, I required larger maps, how could I unlock the rest of the maps (please note: I am still a newbie at Unreal Engine)?

[image]

ishaan95, 2022-09-02:
Have a look at "add assets to a level" [in the CARLA docs here](https://carla.readthedocs.io/en/latest/tuto_M_custom_layers/). It mentions unlocking the level by toggling a lock icon.

glopezdiest, 2022-09-23:
As @ishaan95 said, the docs provided should have all the information needed. Closing this issue as it seems answered but feel free to reopen it if there are still some issues

closed 2022-09-23 by glopezdiest