carla-simulator/carla #5718: Actor.set_location() is relative, but Actor.set_transform() is absolute for actors attached to another actor

Pyrestone, 2022-09-01:
This is kinda part 2 of #857

I would expect both **setters** and **getters** for transform, location, and rotation (btw. why does rotation not have a setter/getter?) to be relative when the actor was spawned attached to another.
with set_location, this works. But with set_transform it doesn't.

Alternatively, adding an argument (called "relative_to" or "parent"), which either takes a carla.Actor or None (default) would make this explicit and allow changing attachments at runtime.

**Why do i need this?** I can't change the rotation of a sensor relative to its Parent at runtime. Translation works

Pyrestone, 2022-10-10:
Correction: the setters are both relative, but the getters are absolute.
Still weird behavior.

Pyrestone, 2022-10-10:
currently, a workaround is to retain the relative transform while spawning, and update via setters based on that.

00bix, 2024-04-26:
I use numpy to multiply matrix versions of the transformations in these cases. As such: 

```

def get_relative_transform(t: carla.Transform, relative_to: carla.Transform) -> carla.Transform:
	rel_m = np.array(relative_to.get_inverse_matrix()).dot(transform_to_matrix(t))
	return matrix_to_transform(rel_m)

def matrix_to_transform(m: np.ndarray) -> carla.Transform:
	return carla.Transform(
		location=carla.Location(x=m[0, 3], y=m[1, 3], z=m[2, 3]),
		rotation=matrix_to_rotation(m[0:3, 0:3])
	)

```
Like that you can get relative transforms which you can use to set the transform of an attached actor. 

I agree that the API could be improved as mentioned before: 
* set_rotation would be nice in company to set_location
* set transform