carla-simulator/carla #5719: Cannot load OpenDRIVE maps using standalone mode example

lpschexn, 2022-09-01:
CARLA version: 0.9.13
Platform/OS: Ubuntu 20.04

I am attempting to load OpenDRIVE maps using the standalone mode. When running the example shown in the [documentation](https://carla.readthedocs.io/en/latest/adv_opendrive/), I get the following error:
```
/opt/carla-simulator/PythonAPI/util$ python3 config.py -x opendrive/TownBig.xodr
load opendrive map 'TownBig.xodr'.
opendrive could not be correctly parsed
```
In addition, this is printed out from my terminal running carla:
```
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
WARNING: Traffic sign 6310 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6282 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6281 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6253 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6244 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6246 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6208 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6207 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6189 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6186 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6182 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6176 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6173 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6172 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6315 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6165 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6162 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6177 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6154 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6149 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6148 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6141 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 5905 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 5893 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6091 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6237 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6096 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6102 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6213 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6178 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6092 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6076 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6181 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6065 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6093 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6056 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6036 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6032 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6019 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6132 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6000 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6071 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6003 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6001 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6007 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6005 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6004 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 5999 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6147 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 5922 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 5965 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6294 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6185 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 5953 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6187 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 5941 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 5960 overlaps a driving lane. Moving out of the road... 
[... 30 lines cut ...]
WARNING: Traffic sign 6272 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6241 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 5791 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 5884 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 5788 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 5790 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6077 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 5920 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 5952 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 5789 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6242 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 5860 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6258 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6184 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6171 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 5769 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6022 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6257 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 6100 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 5817 overlaps a driving lane. Moving out of the road... 
```

These warnings are also seen when I try to load custom maps generated in RoadRunner. 

Any guidance on how to resolve this issue so that the example will successfully load the map would be greatly appreciated. Thank you!

lpschexn, 2022-09-06:
As an update to this issue, I have built CARLA from source and can load maps in the UE4 editor using the same command listed above. However, any advice or guidance on how to get the standalone version to run would be greatly appreciated.

glopezdiest, 2022-09-22:
Hello. I just tested it and I can load TownBig without any issues. The warnings are created when parsing the map, and they are map dependent, regardless of how they are created / used. It is a pretty common warning for maps created in RoadRunner, but it doesn't have a huge impact on the simulation. Have you tried running other xodrs, such as the ones used for Town01 to Town10?

lpschexn, 2022-09-23:
I ended up resolving my errors by building CARLA from source then making a package from the source version. When I tried loading xodr files in that distribution, it loaded up the OpenDRIVE files properly.

I'm still not sure what was keeping me from loading them properly in the version I cloned from the repo, but this

closed 2022-09-23 by lpschexn

TobiMiller, 2022-11-21:
you have installed via deb packages. carla is installed into /opt/ which is owned by root. carla needs to write into this folder. because you cannot run carla server as root, you have to set the owner permission to your user. not a nice solution. but it is working for me.