carla-simulator/carla #5721: make PythonAPI error

zhoupeng0427, 2022-09-03:
I built 0.9.13 as prescribed，but i met a error when i make PythonAPI error.
I only have one version of Python installed（3.8.10）.
please help me.

libcarla.obj : error LNK2001: 无法解析的外部符号 __imp__PyLong_FromUnsignedLongLong
libcarla.obj : error LNK2001: 无法解析的外部符号 _png_set_swap
libcarla.obj : error LNK2001: 无法解析的外部符号 _png_set_invert_mono
libcarla.obj : error LNK2001: 无法解析的外部符号 __imp__PyGILState_Release
libcarla.obj : error LNK2001: 无法解析的外部符号 _png_set_sRGB
libcarla.obj : error LNK2001: 无法解析的外部符号 _png_set_compression_mem_level
libcarla.obj : error LNK2001: 无法解析的外部符号 __imp___Py_NoneStruct
libcarla.obj : error LNK2001: 无法解析的外部符号 _png_get_io_ptr
libcarla.obj : error LNK2001: 无法解析的外部符号 __imp__PyTuple_New
libcarla.obj : error LNK2001: 无法解析的外部符号 __imp__PyLong_FromLongLong
libcarla.obj : error LNK2001: 无法解析的外部符号 __imp__PyFloat_FromDouble
libcarla.obj : error LNK2001: 无法解析的外部符号 _png_create_write_struct
libcarla.obj : error LNK2001: 无法解析的外部符号 "public: unsigned __int64 __thiscall std::hash<struct carla::road::element::Waypoint>::operator()(struct carla::road::element::Waypoint const &)const " (??R?$hash@UWaypoint@element@road@carla@@@std@@QBE_KABUWaypoint@element@road@carla@@@Z)
libcarla.obj : error LNK2001: 无法解析的外部符号 _png_set_gAMA_fixed
libcarla.obj : error LNK2001: 无法解析的外部符号 _png_write_info
libcarla.obj : error LNK2001: 无法解析的外部符号 __imp__PyErr_SetString
libcarla.obj : error LNK2001: 无法解析的外部符号 _png_set_compression_buffer_size
libcarla.obj : error LNK2001: 无法解析的外部符号 __imp__PyLong_FromUnsignedLong
libcarla.obj : error LNK2001: 无法解析的外部符号 _png_set_tIME
libcarla.obj : error LNK2001: 无法解析的外部符号 __imp__PyDict_Type
libcarla.obj : error LNK2001: 无法解析的外部符号 __imp__PyBool_Type
libcarla.obj : error LNK2001: 无法解析的外部符号 __imp__PyFloat_Type
libcarla.obj : error LNK2001: 无法解析的外部符号 _png_write_row
libcarla.obj : error LNK2001: 无法解析的外部符号 __imp__PyLong_Type
libcarla.obj : error LNK2001: 无法解析的外部符号 _png_set_pCAL
libcarla.obj : error LNK2001: 无法解析的外部符号 _png_set_PLTE
libcarla.obj : error LNK2001: 无法解析的外部符号 __imp___Py_Dealloc
libcarla.obj : error LNK2001: 无法解析的外部符号 __imp__PyLong_AsLong
libcarla.obj : error LNK2001: 无法解析的外部符号 __imp__PyImport_AddModule
libcarla.obj : error LNK2001: 无法解析的外部符号 _png_set_sBIT
libcarla.obj : error LNK2001: 无法解析的外部符号 _png_set_oFFs
libcarla.obj : error LNK2001: 无法解析的外部符号 _png_set_filter
libcarla.obj : error LNK2001: 无法解析的外部符号 __imp__PyUnicode_Type
libcarla.obj : error LNK2001: 无法解析的外部符号 _png_set_iCCP
libcarla.obj : error LNK2001: 无法解析的外部符号 __imp__PyObject_IsInstance
libcarla.obj : error LNK2001: 无法解析的外部符号 __imp__PyList_Type
libcarla.obj : error LNK2001: 无法解析的外部符号 _png_set_hIST
libcarla.obj : error LNK2001: 无法解析的外部符号 _png_set_compression_level
libcarla.obj : error LNK2001: 无法解析的外部符号 "public: class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > __thiscall rpc::rpc_error::get_function_name(void)const " (?get_function_name@rpc_error@rpc@@QBE?AV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@XZ)
libcarla.obj : error LNK2001: 无法解析的外部符号 "class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > __cdecl osm2odr::ConvertOSMToOpenDRIVE(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,struct osm2odr::OSM2ODRSettings)" (?ConvertOSMToOpenDRIVE@osm2odr@@YA?AV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V23@UOSM2ODRSettings@1@@Z)
libcarla.obj : error LNK2001: 无法解析的外部符号 _png_destroy_read_struct
libcarla.obj : error LNK2001: 无法解析的外部符号 _png_set_IHDR
libcarla.obj : error LNK2001: 无法解析的外部符号 _png_set_strip_alpha
libcarla.obj : error LNK2001: 无法解析的外部符号 __imp__PyGILState_Ensure
libcarla.obj : error LNK2001: 无法解析的外部符号 _png_set_compression_strategy
libcarla.obj : error LNK2001: 无法解析的外部符号 __imp__PyExc_TypeError
libcarla.obj : error LNK2001: 无法解析的外部符号 _png_set_write_fn
libcarla.obj : error LNK2001: 无法解析的外部符号 __imp__PyObject_IsTrue
libcarla.obj : error LNK2001: 无法解析的外部符号 __imp__PyCallable_Check
libcarla.obj : error LNK2001: 无法解析的外部符号 __imp__PyExc_IndexError
libcarla.obj : error LNK2001: 无法解析的外部符号 _png_create_info_struct
libcarla.obj : error LNK2001: 无法解析的外部符号 __imp__PyMemoryView_FromMemory
libcarla.obj : error LNK2001: 无法解析的外部符号 __imp__PyEval_CallFunction
libcarla.obj : error LNK2001: 无法解析的外部符号 __imp__PyErr_Print
libcarla.obj : error LNK2001: 无法解析的外部符号 __imp__PyGILState_Check
libcarla.obj : error LNK2001: 无法解析的外部符号 _png_set_compression_method
libcarla.obj : error LNK2001: 无法解析的外部符号 _png_write_end
libcarla.obj : error LNK2001: 无法解析的外部符号 __imp__PyObject_Size
libcarla.obj : error LNK2001: 无法解析的外部符号 _png_set_compression_window_bits
libcarla.obj : error LNK2001: 无法解析的外部符号 __imp__PyEval_RestoreThread
libcarla.obj : error LNK2001: 无法解析的外部符号 __imp__PyUnicode_FromStringAndSize
build\lib.win32-3.8\carla\libcarla.cp38-win_amd64.pyd : fatal error LNK1120: 438 个无法解析的外部命令
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x86\\link.exe' failed with exit status 1120

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "E:\UnrealEngine\carla\PythonAPI\carla\dist"!

Into-sight, 2022-09-21:
I have the same problem. Have you solved it？

qbe, 2022-09-21:
pleas quote all code when posting,
like this `inline code`

```
or this
multiline
code block
```

to avoid spamming users with short usernames,

thank you.

zhoupeng0427, 2022-09-22:
> I have the same problem. Have you solved it？

I deleted the files under the build folder, tried many times, and finally succeeded

molua, 2023-04-23:
> > I have the same problem. Have you solved it？
> 
> I deleted the files under the build folder, tried many times, and finally succeeded

which files did you delete?  I have the same problem

YeZibn, 2024-12-25:
I changed the python version from 32-bit to 64-bit and it worked

qingtong1215, 2025-09-17:
I also encountered this issue, so here's a complete record of what I tried: I uninstalled Anaconda, then uninstalled all other versions of Python. I reinstalled Python 3.8.10, then deleted the folders named with "-install" in CARLA\Build. I also deleted everything under the carla\PythonAPI\carla\build directory. After that, I ran make clean and make PythonAPI in the carla directory, and the error no longer appeared.

Blyron, 2026-11-12:
This is an error from the old Make/`.bat` build of CARLA 0.9.x on Unreal Engine 4. That build system is gone: CARLA 1.0 (Unreal Engine 5.8, branch `ue58-dev`) builds with CMake, which downloads and builds the dependencies itself. See [Build on Windows](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/build_windows_ue5.md). Closing as obsolete. If the current build fails for you, please open a new issue with the full log.

closed 2026-11-12 by Blyron